P0cL4bs/WiFi-Pumpkin-deprecated #285: WiFi-Pumpkin v08.5 running on a raspberry pi 3, no Internt on the AP

harames2001, 2017-10-13:
Kali linux is running on an raspberry pi 3.(kali-2.1.2-rpi2, +kali-linux-top10, +kali-linux-wireless)
Then i installed WiFi-Pumpkin:
1.sudo apt-get update
2.pip install service_identity
3.pip install scapy_http
4.sudo apt-get install mitmproxy
5.git clone https://github.com/P0cL4bs/WiFi-Pumpkin.git
6.cd WiFi-Pumpkin
7.chmod +x installer.sh
8../installer.sh --install
9.wifi-pumpkin

So WiFi pumking is running, i am connected to the internet via eth0.
For the AP i use: AWUS036NHA
I only switchend the network adapter to the USB-Wirelesscard (wlan1).
All other settings are standart.

Now i "START" AP. On the bottom AP-Startus is ON.

I use a other Laptop to conect to the WIFI, but there is no internet connection over the AP!

THX for helping

mh4x0f, 2017-10-14:
start the WP again and send me the ouput of command bellow.
``` sh
sudo iptables -t nat -L
```

harames2001, 2017-10-14:
root@kali:~# sudo iptables -t nat -L && ifconfig
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination
REDIRECT   tcp  --  anywhere             anywhere             tcp dpt:http redir                          ports 8080

Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination
MASQUERADE  all  --  anywhere             anywhere
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.0.11  netmask 255.255.255.0  broadcast 10.0.0.255
        inet6 fe80::ba27:ebff:fe92:3db0  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:92:3d:b0  txqueuelen 1000  (Ethernet)
        RX packets 68359  bytes 31590203 (30.1 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 68236  bytes 62347057 (59.4 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 0  (Local Loopback)
        RX packets 39790  bytes 59583515 (56.8 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 39790  bytes 59583515 (56.8 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1400
        ether bc:f6:85:03:36:5b  txqueuelen 1000  (Ethernet)
        RX packets 7  bytes 440 (440.0 B)
        RX errors 0  dropped 3  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1400
        inet 10.0.0.1  netmask 255.0.0.0  broadcast 10.255.255.255
        ether bc:f6:85:03:36:5b  txqueuelen 1000  (Ethernet)
        RX packets 149  bytes 12007 (11.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 24  bytes 3242 (3.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

..........................................................................................................................................................................................................
+WP use wlan1
+one device is connectet to the AP(no Internt on the "victim")

mh4x0f, 2017-10-15:
ok, now you need to change the IP address classes to **B** or **C** (settings -> Class Ranges) and save settings, you have choose the Class range different of your network. I'll add the code for check this problem coming soon. 
[image]

harames2001, 2017-10-15:
OK now it works, THX

closed 2017-10-15 by harames2001

curtismany, 2018-07-29:
I see that u already got the solution, congrats could anyone help me? when I have al set up I just get the error wlan0 interface not found and I can connect with my phone and surf on the web but I get no passwords when I track from the kali. I have the same error as this guy:
https://github.com/P0cL4bs/WiFi-Pumpkin/issues/390
Any help would be appreciated, thanks.

Rdgeno, 2019-09-13:
Having same problem no connection on AP. Tried everything. Gui shows no problem and connected. No internet.