P0cL4bs/WiFi-Pumpkin-deprecated #288: WiFi-Pumpkin v08.5 running on a raspberry pi 3, no Internt on the AP using Responder

harames2001, 2017-10-20:
Kali linux is running on an raspberry pi 3.(kali-2.1.2-rpi2, +kali-linux-top10, +kali-linux-wireless)

So WiFi pumking is running, i am connected to the internet via eth0.
For the AP i use: AWUS036NHA

I switchend the network adapter to the USB-Wirelesscard (wlan1).
DHCP-Settings: Class-B-Adress
SSLstrip+Dns2proxy is enabled
When i start the AP now, there is a internet connection to the victim.
So far so good.

Now i start "Responder" (in Settings > save settigs  and in Plugins)

There is no internt connection on the victim.

What I am doing wrong?

mh4x0f, 2017-10-23:
the responder plugin start the server on port 53, this do the AP not share internet. for solve this go to in Plugins > Responder row -> click button change, now change **DNS** option for **Off** and click in save settings as you can see in screenshot bellow. 
[image]

closed 2017-10-26 by mh4x0f

ngrok01, 2019-03-20:
I get this message when I try to connect from a device to the PA "unable to join the network"