P0cL4bs/WiFi-Pumpkin-deprecated #289: Empty Window ERROR

Nattas, 2017-10-21:
## What's the problem (or question)?

Window Is Empty! Tried All The Things Suggested Such As: *export QT_X11_NO_MITSHM=1*
And Reinstalling.

Log:
~$ sudo wifi-pumpkin
Loading GUI...
WiFi-Pumpkin Running!
X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    130 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x13f
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 5 (X_ShmCreatePixmap)
  Resource id:  0x3400018
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x3400019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x3400019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x3400019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x3400019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x3400019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x3400019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x3400019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x3400019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x3400019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x3400019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x3400019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x3400019
*and it goes on...*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): doesnt matter
* Version used tool: 0.8.5
* Virtual Machine (yes or no and which): no
* Operating System and version: Kali Linux 2017.1 (2017.3.2)

closed 2017-10-23 by mh4x0f

Nattas, 2017-10-28:
Whats The Fix????

d4rkcon, 2017-11-26:
Try: sudo QT_X11_NO_MITSHM=1 wifi-pumpkin

arno257801, 2018-01-05:
Adding in the file 
sudo nano /etc/environment

Prompt: sudo QT_X11_NO_MITSHM=1

and saving it helped me :) 

If it doesn't use 
Prompt: xhost +

and try again. Hope it will work.