P0cL4bs/WiFi-Pumpkin-deprecated #290: dhcp can't assign  ip

smatiush, 2017-10-22:
## What's the problem (or question)?

in settings i set isc-dhcp-server but when a device try a connection this can't get a ip.
if i set a python-dhcp all work fine but i can't see any connected device.

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):(D-link)DWA-140
* Version used tool:0.8.5
* Virtual Machine (yes or no and which):no
* Operating System and version: kali-rolling

Lexus89, 2017-10-29:
I get the following in the debug for DHCP when running as root. Running VM ubuntu, latest version pumpkin. Also isc-dhcp server does not work. Strangely it also kills my uplink to the internet when starting the tool.

```
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/packets/dhcpserver.py", line 432, in run
    self.dhcp_ack(message)
  File "/usr/share/WiFi-Pumpkin/core/packets/dhcpserver.py", line 401, in dhcp_ack
    self.sock.sendto(response, (self.broadcast, 68))
socket.error: [Errno 1] Operation not permitted
```

mh4x0f, 2017-10-29:
@Lexus89  this is socket.error when try bind the port 68. this means that there is another process running in port 68, probably the isc-dhcp is return same error. you can kill the process by port, just use the google "how kill a process running on particular port in Linux?"

> Strangely it also kills my uplink to the internet when starting the tool.

this is normal, the tool is configuring the adapter wireless.

Lexus89, 2017-10-30:
I seem to be unable to get an IP when using python DHCP hmm

closed 2017-11-05 by mh4x0f

kwrnel, 2017-11-07:
Não se esqueçam que pra escutar em porta baixa, tem que estar logado como super usuário. ;-)