P0cL4bs/WiFi-Pumpkin-deprecated #292: Failed Building Cryptography

jeretc, 2017-10-24:
## What's the problem (or question)?

*installation error:
failed building cryptography

error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

Can't rollback cryptography, nothing uninstalled.
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-iebK7s/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-yUR0Ix-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-iebK7s/cryptography/
*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):TL-WN722N
* Version used tool: 0.8.5
* Virtual Machine (yes or no and which): no
* Operating System and version: parrotsec 3.8

closed 2017-10-25 by mh4x0f

mh4x0f, 2017-10-25:
I don't solve any problem by module, just use google my friend

jeretc, 2017-10-25:
Yes, been trying to find a solution past 1 week before i post it here . .
 :(

Hopefully i'l find one soon.. thanks.

jeretc, 2017-10-28:
@mh4x0f , this is some updates from mitmproxy support .. just thought you might want to know. I still could not get it to work.

It seems you are trying to install mitmproxy 0.18.2 - which is very old and unfortunately not supported any more (and it is python2.7).

Please use mitmproxy 2.0.2 and Python 3.5 (or higher).
Make sure to use pip3 install mitmproxy to get the correct version installed.