P0cL4bs/WiFi-Pumpkin-deprecated #296: Redirect Traffic from all domain doesn't working

Thenatdead, 2017-11-02:
## Redirect Traffic from all domain doesn't working ?
Hi, i use DNS Spoofer with Windows Update Attack Generator.
In  DNS Spoofer i activate "Redirect traffic from all domains" but when on AP Wifi i go on any website i have no connection screen but if i disable "Redirect traffic from all domains" and add manualy "google.com" its working on this and only on this.
Thanks for helping me (screen of my tool config)
[image]

#### 
* Card wireless adapters name : Don't know the name but working good.
* Version used tool: Last Update
* Virtual Machine : No
* Operating System and version: Kali Linux 2017.2

ssrmpc, 2017-11-02:
Since 99% of all the issues I had were related to the Wifi-Adapter or better the related driver, I Kindly ask you to provide us the adapter name and chipset. Do you use an external USB Wifi Adapter such as one from e.g. Alfa AWUS036ACH (RTL8812AU)

Was your adapter driver supported  in the Linux Kernel? Did you have to install the driver via apt-get install realtek-rtl88xxdau-dkms?
Does your adapter support monitor mode?
Does your adapter support AP?

Well then... talk to you later

Thenatdead, 2017-11-02:
Its pw-dn4210D USB.My adapter support monitor and AP.

Thenatdead, 2017-11-02:
Chipset: Atheros AR9271

ssrmpc, 2017-11-03:
Good news.... you shouldn't have any issue with this driver (I know... you
told me.... but seeing is believing :-)

*Something for the Pumpking-Specialist:* Can the wlan1 interface be used in
that many processes?

Anyway.... I try to reproduce your error after I had a good sleep

Cheers
-Marc

Thenatdead, 2017-11-03:
Someone have solution please ?

mh4x0f, 2017-11-28:
bad news, I'm work for solve this bug, but I see in old report on python dev [issue socketserver](https://bugs.python.org/issue14574)  that problem not  has been  solve. I'm find another way for redirect the client to web page attack without use the iptables. case not to be possible i'll implement another python web server.

mh4x0f, 2017-12-06:
@Thenatdead  I started new implementation for create the web server using own module **SimpleHTTPServer**. anyway, the python web server is very slow, but will work normally.

closed 2017-12-15 by mh4x0f

donniemillan, 2018-04-05:
This issue still appears to exist. I can redirect using a specific domain such as google.com, but redirecting all domains doesn't work.

I'm using an Alfa AWUS051NH.