P0cL4bs/WiFi-Pumpkin-deprecated #299: Can’t connect to AP

bigj75024, 2017-11-10:
I am able to get my AP up and running and everything looks good. The network is available in the list of networks, however, when attempting to connect, I’m unsuccessful. I’ve tried connecting to the AP from Windows, Android, and iOS, all to no avail. 

*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): alfa awus036nh v2
* Version used tool: 
* Virtual Machine (yes or no and which): yes, virtualbox
* Operating System and version: Kali 2017.2, kernel 4.12

mh4x0f, 2017-11-13:
look this comment https://github.com/P0cL4bs/WiFi-Pumpkin/issues/247#issuecomment-312542569  if doesn't work try this #285.

bigj75024, 2017-11-14:
> look this comment #247 (comment) if doesn't work try this #285.

I still have the same issue. The AP is up, but no devices can connect.

mh4x0f, 2017-11-14:
i don't find any solution for now, but I'll install the virtualbox and make some tests. I recommend to use VMware

bigj75024, 2017-11-14:
I'm installing your app on a standalone Kali box now. I'm in hopes that
makes a difference. Thanks for your help up to this point.

bigj75024, 2017-11-14:
Ok, good news! I've installed wifi-pumpkin onto a standalone kali box, and I'm now able to connect to the AP without any issues, so it does in fact look as if virtualbox is causing the problem in this case. However, while I'm able to connect, I have no internet access, but I am using the Alfa card, as well as the internal wireless card, so I'm going to modify the configuration for IP tables, and see how that works.

bigj75024, 2017-11-14:
Done. We are in business now. Thanks for everything.

closed 2017-11-14 by bigj75024

GlenDeHahn, 2019-03-29:
I can not get this application to work in virtul box either... I have tried everything. you cant connect to the AP no matter how you adjust the settings.. did you ever figure out a fix for this?

Sh4kim, 2019-04-12:
Hello, 
  I had the same problem, and thanks to bigj75024 I was able to understand what happen for me.
I was using an usb 2.0 version, that don't work well with Kali or maybe my wifi adaptator. I changed the usb version into 1.1 and 3.0 and both work well. This is in the "USB" menu in the configuration pannel.