P0cL4bs/WiFi-Pumpkin-deprecated #304: WiFi-Pumpkin need PyQt4 :(

SamuX99, 2017-11-25:
My problem is that when I start wifi-pumpkin I get the error "WiFi-Pumpkin need PyQt4 :(".
I have already tried to reinstall it, and also reinstall python, without results.
Could someone give me a solution?
I use kali linux.

SamuX99, 2017-11-25:
Why you close my issue? @mh4x0f

mh4x0f, 2017-11-28:
#282 look

rootsina, 2017-11-30:
and i have still problem (same)
root@kali:~# sudo wifi-pumpkin
WiFi-Pumpkin need PyQt4 :(
root@kali:~#

SamuX99, 2017-11-30:
@mh4x0f Im agree, but I had already searched and I have not found any solution. 
That's why I opened the issue, in case someone puts a reference or has already solved the problem...
Thanks

Chenxinjian, 2017-12-02:
wifi-pumpkin.py has this problem.

from core.utility.application import ApplicationLoop,QtGui
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "core/utility/application.py", line 3, in <module>
    from core.main import version
  File "core/main.py", line 28, in <module>
    from core.widgets.tabmodels import (
  File "core/widgets/tabmodels.py", line 6, in <module>
    from core.utility.threads import ThreadPopen
  File "core/utility/threads.py", line 19, in <module>
    from core.servers.proxy.http.controller.handler import MasterHandler
  File "core/servers/proxy/http/controller/handler.py", line 4, in <module>
    from mitmproxy import controller,flow
  File "/usr/lib/python2.7/dist-packages/mitmproxy/flow/__init__.py", line 5, in <module>
    from mitmproxy.flow.master import FlowMaster
  File "/usr/lib/python2.7/dist-packages/mitmproxy/flow/master.py", line 15, in <module>
    from mitmproxy.onboarding import app
  File "/usr/lib/python2.7/dist-packages/mitmproxy/onboarding/app.py", line 10, in <module>
    from mitmproxy.proxy import config
  File "/usr/lib/python2.7/dist-packages/mitmproxy/proxy/__init__.py", line 4, in <module>
    from .root_context import RootContext
  File "/usr/lib/python2.7/dist-packages/mitmproxy/proxy/root_context.py", line 10, in <module>
    from mitmproxy import protocol
  File "/usr/lib/python2.7/dist-packages/mitmproxy/protocol/__init__.py", line 40, in <module>
    from .tls import TlsClientHello
  File "/usr/lib/python2.7/dist-packages/mitmproxy/protocol/tls.py", line 11, in <module>
    from mitmproxy.contrib.tls import _constructs
  File "/usr/lib/python2.7/dist-packages/mitmproxy/contrib/tls/_constructs.py", line 7, in <module>
    from construct import (Array, Bytes, Struct, UBInt16, UBInt32, UBInt8, PascalString, Embed, TunnelAdapter, GreedyRange,
ImportError: cannot import name UBInt16

Chenxinjian, 2017-12-02:
I had solve the problem . @SamuX99 just download construct-2.5.2-py2.py3-none-any.whl and install it.

mh4x0f, 2017-12-02:
@Chenxinjian create a new issue and comment the solution for another users thanks.

gussj, 2017-12-04:
I have the same problema and i can´t fix it. Tried the construct but didn´t work.

closed 2017-12-07 by mh4x0f