P0cL4bs/WiFi-Pumpkin-deprecated #306: WiFi-Pumpkin need PyQt4 - Last update in Kali

gussj, 2017-12-04:
I tried other solutions and nothing works, the message keeps appearing.

Beside that i tried to install again and i have this message when Building wheel for cryptography

Running setup.py install for cryptography ... error
Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;file='/tmp/pip-build-e1F_wl/cryptography/setup.py';f=getattr(tokenize, 'open', open)(file);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, file, 'exec'))" install --record /tmp/pip-LpNdB6-record/install-record.txt --single-version-externally-managed --compile:
running install
running build
running build_py
creating build
creating build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/cryptography
copying src/cryptography/utils.py -> build/lib.linux-x86_64-2.7/cryptography
copying src/cryptography/init.py -> build/lib.linux-x86_64-2.7/cryptography
copying src/cryptography/exceptions.py -> build/lib.linux-x86_64-2.7/cryptography
copying src/cryptography/fernet.py -> build/lib.linux-x86_64-2.7/cryptography
copying src/cryptography/about.py -> build/lib.linux-x86_64-2.7/cryptography
creating build/lib.linux-x86_64-2.7/cryptography/hazmat
copying src/cryptography/hazmat/init.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat
creating build/lib.linux-x86_64-2.7/cryptography/x509
copying src/cryptography/x509/oid.py -> build/lib.linux-x86_64-2.7/cryptography/x509
copying src/cryptography/x509/init.py -> build/lib.linux-x86_64-2.7/cryptography/x509
copying src/cryptography/x509/base.py -> build/lib.linux-x86_64-2.7/cryptography/x509
copying src/cryptography/x509/name.py -> build/lib.linux-x86_64-2.7/cryptography/x509
copying src/cryptography/x509/general_name.py -> build/lib.linux-x86_64-2.7/cryptography/x509
copying src/cryptography/x509/extensions.py -> build/lib.linux-x86_64-2.7/cryptography/x509
creating build/lib.linux-x86_64-2.7/cryptography/hazmat/backends
copying src/cryptography/hazmat/backends/interfaces.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends
copying src/cryptography/hazmat/backends/init.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends
copying src/cryptography/hazmat/backends/multibackend.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends
creating build/lib.linux-x86_64-2.7/cryptography/hazmat/bindings
copying src/cryptography/hazmat/bindings/init.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/bindings
creating build/lib.linux-x86_64-2.7/cryptography/hazmat/primitives
copying src/cryptography/hazmat/primitives/init.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/primitives
copying src/cryptography/hazmat/primitives/serialization.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/primitives
copying src/cryptography/hazmat/primitives/keywrap.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/primitives
copying src/cryptography/hazmat/primitives/hashes.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/primitives
copying src/cryptography/hazmat/primitives/hmac.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/primitives
copying src/cryptography/hazmat/primitives/padding.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/primitives
copying src/cryptography/hazmat/primitives/cmac.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/primitives
copying src/cryptography/hazmat/primitives/constant_time.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/primitives
creating build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/commoncrypto
copying src/cryptography/hazmat/backends/commoncrypto/ciphers.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/commoncrypto
copying src/cryptography/hazmat/backends/commoncrypto/init.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/commoncrypto
copying src/cryptography/hazmat/backends/commoncrypto/hashes.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/commoncrypto
copying src/cryptography/hazmat/backends/commoncrypto/hmac.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/commoncrypto
copying src/cryptography/hazmat/backends/commoncrypto/backend.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/commoncrypto
creating build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/openssl
copying src/cryptography/hazmat/backends/openssl/utils.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/openssl
copying src/cryptography/hazmat/backends/openssl/ec.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/openssl
copying src/cryptography/hazmat/backends/openssl/ciphers.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/openssl
copying src/cryptography/hazmat/backends/openssl/init.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/openssl
copying src/cryptography/hazmat/backends/openssl/decode_asn1.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/openssl
copying src/cryptography/hazmat/backends/openssl/hashes.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/openssl
copying src/cryptography/hazmat/backends/openssl/hmac.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/openssl
copying src/cryptography/hazmat/backends/openssl/cmac.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/openssl
copying src/cryptography/hazmat/backends/openssl/backend.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/openssl
copying src/cryptography/hazmat/backends/openssl/dsa.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/openssl
copying src/cryptography/hazmat/backends/openssl/encode_asn1.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/openssl
copying src/cryptography/hazmat/backends/openssl/x509.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/openssl
copying src/cryptography/hazmat/backends/openssl/rsa.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/backends/openssl
creating build/lib.linux-x86_64-2.7/cryptography/hazmat/bindings/commoncrypto
copying src/cryptography/hazmat/bindings/commoncrypto/init.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/bindings/commoncrypto
copying src/cryptography/hazmat/bindings/commoncrypto/binding.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/bindings/commoncrypto
creating build/lib.linux-x86_64-2.7/cryptography/hazmat/bindings/openssl
copying src/cryptography/hazmat/bindings/openssl/init.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/bindings/openssl
copying src/cryptography/hazmat/bindings/openssl/_conditional.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/bindings/openssl
copying src/cryptography/hazmat/bindings/openssl/binding.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/bindings/openssl
creating build/lib.linux-x86_64-2.7/cryptography/hazmat/primitives/kdf
copying src/cryptography/hazmat/primitives/kdf/concatkdf.py -> build/lib.linux-x86_64-2.7/cryptography/hazmat/primitives/kdf
[... 450 lines cut ...]
^~~~~~~~~~~~~~~~~~~
build/temp.linux-x86_64-2.7/_openssl.c: In function ‘_cffi_f_X509_get0_signature’:
build/temp.linux-x86_64-2.7/_openssl.c:59549:25: warning: passing argument 1 of ‘X509_get0_signature’ from incompatible pointer type [-Wincompatible-pointer-types]
{ X509_get0_signature(x0, x1, x2); }
^~
In file included from /usr/include/openssl/cms.h:16:0,
from build/temp.linux-x86_64-2.7/_openssl.c:532:
/usr/include/openssl/x509.h:552:6: note: expected ‘const ASN1_BIT_STRING ** {aka const struct asn1_string_st **}’ but argument is of type ‘ASN1_OCTET_STRING ** {aka struct asn1_string_st **}’
void X509_get0_signature(const ASN1_BIT_STRING **psig,
^~~~~~~~~~~~~~~~~~~
build/temp.linux-x86_64-2.7/_openssl.c:59549:29: warning: passing argument 2 of ‘X509_get0_signature’ from incompatible pointer type [-Wincompatible-pointer-types]
{ X509_get0_signature(x0, x1, x2); }
^~
In file included from /usr/include/openssl/cms.h:16:0,
from build/temp.linux-x86_64-2.7/_openssl.c:532:
/usr/include/openssl/x509.h:552:6: note: expected ‘const X509_ALGOR ** {aka const struct X509_algor_st **}’ but argument is of type ‘X509_ALGOR ** {aka struct X509_algor_st **}’
void X509_get0_signature(const ASN1_BIT_STRING **psig,
^~~~~~~~~~~~~~~~~~~
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

----------------------------------------
Can't rollback cryptography, nothing uninstalled.
Command "/usr/bin/python -u -c "import setuptools, tokenize;file='/tmp/pip-build-e1F_wl/cryptography/setup.py';f=getattr(tokenize, 'open', open)(file);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, file, 'exec'))" install --record /tmp/pip-LpNdB6-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-e1F_wl/cryptography/

Thanks!

4p3rtur3, 2017-12-04:
Confirmed from my side.
I have exactly the same problem!
Dont know how to fix that.

Kind regards.

mh4x0f, 2017-12-04:
I don't solve any problem about module

closed 2017-12-04 by mh4x0f

littleheary, 2018-11-23:
i also have this problem,how to solve?