P0cL4bs/WiFi-Pumpkin-deprecated #308: WiFi-Pumpkin need PyQt4 :(

Spoooof, 2017-12-11:
## What's the problem (or question)?
I am getting this problem with kali 2017.3 (Debian version Version 3.26.2)

"WiFi-Pumpkin need PyQt4 :("

When "sudo wifi-pumpkin"

I have tried installing every possible thing anyone have mentioned without luck. If I try starting wifi pumpkin from "app drawer" nothing happen. I have installed wifi-pumpkin once before with luck and I installed it the same way this time so I know it should work but it ain't. I do not remember what kali version I used earlier but this problem arise when I did a fresh install of kali linux. I have done apt-get update, apt-get upgrade and all of those, I have tried reinstalling lastest kali release, rebooting my computer etc.. I won't work

*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): ALFA AWUS036NHA
* Version used tool: 0.8.5
* Virtual Machine (yes or no and which): YES, VMware Fusion
* Operating System and version: Mac OS X 10.13.2 (High Sierra HOST) Kali rolling 2017.3

LexLinux, 2017-12-11:
I have the same issue. I've also installed the latest construct-2.5.2 as above.
my card is the ALFA AWUS036NHA
Version 0.8.5
Virtual machine (12.5.7 build-5813279) running on Windows 10 Pro 1703
Kali Linux Rolling 2017.3

Showing errors : 
`Can't rollback cryptography, nothing uninstalled.
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-VC9xpj/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-1LhaVq-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-VC9xpj/cryptography/`
'Failed building wheel for cryptography'

closed 2017-12-11 by mh4x0f

mh4x0f, 2017-12-11:
try again download the tool upgraded,probably you can see what's causing the error. remembering that I do not solve any bug installing modules. google is your friend

LexLinux, 2017-12-11:
Thank you, I will try again

commit ffa2c923 (2017-12-11): fixed #308