P0cL4bs/WiFi-Pumpkin-deprecated #309: ImportError: No module named mitmproxy.models

guilhermebasalia, 2017-12-12:
After you run the sudo wifi-pumpkin displays the following error.

Traceback (most recent call last):
  File "wifi-pumpkin.py", line 37, in <module>
    from core.utility.application import ApplicationLoop
  File "/usr/share/WiFi-Pumpkin/core/utility/application.py", line 3, in <module>
    from core.main import version
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 28, in <module>
    from core.widgets.tabmodels import (
  File "/usr/share/WiFi-Pumpkin/core/widgets/tabmodels.py", line 6, in <module>
    from core.utility.threads import ThreadPopen
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 19, in <module>
    from core.servers.proxy.http.controller.handler import MasterHandler
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/http/controller/handler.py", line 1, in <module>
    from plugins.extension import *
  File "/usr/share/WiFi-Pumpkin/plugins/extension/beef.py", line 1, in <module>
    from mitmproxy.models import decoded
ImportError: No module named mitmproxy.models
* Operating System and version: 

I'm using ParrotOS.

LexLinux, 2017-12-12:
Same issue here.

abdozekaa, 2017-12-12:
Same problem with kali nethunter .. waiting

wimdecorte, 2017-12-13:
Same issue with Kali on Raspberry Pi

7rav6n, 2017-12-13:
Same error here... dosen't work on Kali 2017.3 full install, or Raspberry Pi3 (Kali).

usama7628674, 2017-12-13:
Use these commands and see it that helps for anyone of you
`pip install -U cryptography`
`pip3 install mitmproxy`

guilhermebasalia, 2017-12-13:
The error persists even after you have installed the packages.

└──╼ $sudo pip install -U cryptograph
Successfully installed asn1crypto-0.23.0 cryptography-2.1.4 idna-2.6 ipaddress-1.0.18
└──╼ $sudo pip3 install mitmproxy
Requirement already satisfied: mitmproxy in /usr/lib/python3/dist-packages

usama7628674, 2017-12-13:
I'm also getting `no module named mitmproxy.models`.The problem is after updating to kali linux 2016.3 mitmproxy is also updated.Now when you run `pip install mitmproxy` it uses cryptography-1.5.3 version and that fails to build wheels for it.mitmproxy 0.18.2 is the last version that supports both python 2 and 3.So mitmproxy installs successfully for python 3 using this command `pip3 install mitmproxy` but it doesn't install for python 2 using `pip install mitmproxy`.I've spent a day to rsolve this problem but couldn't do so.I'm thinking that how can I install different version of cryptography?

wimdecorte, 2017-12-13:
Same as @guilhermebasalia - error persists and mitmproxy is already there.

usama7628674, 2017-12-13:
It has something to do with openssl,using different versions of openssl throw different errors and warnings.

usama7628674, 2017-12-13:
Tried mitmproxy==0.17 but still doesn't install.

usama7628674, 2017-12-13:
Can anyone try installing mitmproxy on kali 2016.2?

7rav6n, 2017-12-13:
Tried clean install on Kali 2017.2 on Raspberry Pi3, but still no luck.

usama7628674, 2017-12-14:
I'll check on kali 2017.3 (live) and report here.

drjohannspector, 2017-12-14:
Tried on kali 2017.3 on a live boot, does not work. fresh install too.

usama7628674, 2017-12-14:
Ok.I tried kali 2017.3 on a live boot and now wifi-pumpkin works.

7rav6n, 2017-12-14:
It really works on Kali  2017.3 live boot? It didnt work in my case.(live persistent).

usama7628674, 2017-12-14:
@7rav6n Yes,it really is working for me.Try live without persistence.Just select first choice that says live when booting.

usama7628674, 2017-12-14:
@7rav6n @drjohannspector @wimdecorte @guilhermebasalia @abdozekaa Ok I resolved the issue.Just live boot kali 2017.3 **(no persistence)** and copy all the items from `/usr/lib/python2.7/dist-packages` and paste into your existing kali (I'm assuming you're using dual boot) same directory `/usr/lib/python2.7/dist-packages`.That's how I resolved this issue.

wimdecorte, 2017-12-14:
I'll give it a shot but mine is not a dual boot, it's Kali running on a Raspberry Pi.

wimdecorte, 2017-12-14:
Do we need to install pumpkin on the live boot first, or just grab what's there on a plain vanilla live boot?

usama7628674, 2017-12-14:
No you don't need to install wifi-pumpkin

7rav6n, 2017-12-14:
@usama7628674 Ok, it does work this way on Kali 2017.3, thanks for solution. And is there  way to make it work on Raspberry Pi3 ?

drjohannspector, 2017-12-15:
Now I am seeing this error when starting. Did not get any errors during the install after performing the steps above. Anyone have any ideas for me? It would be greatly appreciated. 

Traceback (most recent call last):
  File "wifi-pumpkin.py", line 37, in <module>
    from core.utility.application import ApplicationLoop
  File "/usr/share/WiFi-Pumpkin/core/utility/application.py", line 3, in <module>
    from core.main import version
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 24, in <module>
    from core.utils import (
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 8, in <module>
    import netifaces
ImportError: No module named netifaces

drjohannspector, 2017-12-15:
Made sure to copy the entire directory and its contents, with chmod a+x on /dist-packages/

copied with no errors, not sure what I did wrong but this time this is my output.

root@kali:~/WiFi-Pumpkin# sudo wifi-pumpkin
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 35, in <module>
    from core.loaders.checker.depedences import check_dep_pumpkin
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/depedences.py", line 2, in <module>
    from core.utility.collection import SettingsINI
  File "/usr/share/WiFi-Pumpkin/core/utility/collection.py", line 2, in <module>
    from PyQt4.QtCore import QSettings
ImportError: /usr/lib/python2.7/dist-packages/sip.x86_64-linux-gnu.so: file too short

usama7628674, 2017-12-15:
@7rav6n I don't have raspberry pi so I'm not familiar with arm platform and not sure how many packages it comes preinstalled.Looking at the size of size of arm of just 0.6GB I think it comes with small number of packages installed.Do what I said above and see if that resolves your problem.

usama7628674, 2017-12-15:
@drjohannspector just install this and report back
`apt-get install python-sip`

7rav6n, 2017-12-15:
Tried diffrent things but still no luck with arm (rp3).I did install diffrent packages, i checked every solution i could find in google. If anyone manages to make it work please post back... i run out of ideas.

mockedxe, 2017-12-15:
what happens is a solution. PLEASE..

mh4x0f, 2017-12-15:
due to large quantity of users can't install the module mitmproxy, I 'll disable the plugin mitmproxy and set another as default. this will solve the problem temporarily. wait ....

mockedxe, 2017-12-16:
What time

mockedxe, 2017-12-17:
I'm crazy when I try to work. In 2017.2, WiFi Pumpkins were working. I made a clean install with 2017.3 and now I will not work.

There seems to be a problem with cryotography and mitmproxy.modules

I'm running Kali Linux 2017.3 in VMWare
Can you help me. Please 😇

maxitis92, 2017-12-18:
[image]

i have this problem. how can i solve it?

Patxe, 2017-12-19:
@usama7628674 Ok, it does work this way on Kali 2017.3. 
Thxs for solution.

mockedxe, 2017-12-19:
@Patxe 
I'm crazy when I try to work. In 2017.2, WiFi Pumpkins were working. I made a clean install with 2017.3 and now I will not work.

There seems to be a problem with cryotography and mitmproxy.modules

I'm running Kali Linux 2017.3 in VMWare
Can you help me. Please 😇

usama7628674, 2017-12-20:
@mockedxe Read my post above.It's the only solution right now.

mockedxe, 2017-12-20:
@usama7628674
please re-write

mockedxe, 2017-12-20:
@usama7628674 this is the solution ? 👍
just install this and report back
apt-get install python-sip
also install these packages to check if they exist in your machine or not
apt-get install python-qscintilla2 python-qt4 python-qt4-sql python-sip

if netifacesis not installed then do this
pip install netifaces

usama7628674, 2017-12-20:
@mockedxe No,check my 9th post.

mockedxe, 2017-12-20:
@usama7628674 Thanks bro.

mockedxe, 2017-12-20:
But 
This is /usr/lib/python2.7/dist-packages
where do we copy.

usama7628674, 2017-12-20:
@mockedxe Are you using dual boot or live boot?

mockedxe, 2017-12-20:
@usama7628674 i am using  VMware

mockedxe, 2017-12-20:
mitmproxy not download
 Running setup.py bdist_wheel for cryptography ... - gives an error here

usama7628674, 2017-12-20:
@mockedxe live boot kali (using usb) and install wifi-pumpkin.

mockedxe, 2017-12-20:
@usama7628674  wifi pumpkin Why does not VMware work?

usama7628674, 2017-12-20:
@mockedxe Did it work on live usb?

mockedxe, 2017-12-20:
@usama7628674 I did not want to try.
because there is no need for just one program

mockedxe, 2017-12-20:
@usama7628674  There is no other way?

LTE-M, 2017-12-21:
@maxitis92  same problem  Have you solved it?

mockedxe, 2017-12-21:
@maxitis92 no!!

Esser420, 2017-12-23:
My project also uses mitmproxy and I was facing the same issue recently.
My solution was deleteing everything cryptography related and also mitmproxy and then reinstalling it, the needed version is 0.18.2. This and also my project need an update to support python3 in order to use the new mitmproxy. The quick fix I used was:

```
apt-get remove --purge python-cryptography
pip uninstall mitmproxy
pip3 uninstall mitmproxy
pip install -U cryptography
pip install mitmproxy==0.18.2
```

mockedxe, 2017-12-23:
@Esser420  VMware using do you?

Esser420, 2017-12-23:
@mockedxe

No Kali linux on bare metal.

mockedxe, 2017-12-23:
@Esser420
Kali Linux bare metal how to. which videos have. link to throw you?

mockedxe, 2017-12-23:
@Esser420
I'm crazy when I try to work. In 2017.2, WiFi Pumpkins were working. I made a clean install with 2017.3 and now I will not work.

There seems to be a problem with cryotography and mitmproxy.modules

I'm running Kali Linux 2017.3 in VMWare
Can you help me. Please 😇

mockedxe, 2017-12-23:
@Esser420
Do you know?

Esser420, 2017-12-23:
@mockedxe
Bare metal just means I installed it directly on my machine instead of a VM. I do not recommend this setup however.. No links to show you, just try running those commands I show.

mockedxe, 2017-12-23:
@Esser420
Thanks bro.

Esser420, 2017-12-23:
@mockedxe

Did it work tho?

warriordz, 2017-12-27:
why wifi-pumpkin not orking in kali linux 2017.3 and what is soultion

[125 more replies]

closed 2017-12-29 by mh4x0f