P0cL4bs/WiFi-Pumpkin-deprecated #314: wifi-pumpkin installaton error = scapy http

root1825, 2017-12-29:
## What's the problem (or question)?

WHenever i try to start this....... ;( please help
root@kali:wifi-pumpkin
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 37, in <module>
    from core.utility.application import ApplicationLoop
  File "/usr/share/WiFi-Pumpkin/core/utility/application.py", line 3, in <module>
    from core.main import version
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 28, in <module>
    from core.widgets.tabmodels import (
  File "/usr/share/WiFi-Pumpkin/core/widgets/tabmodels.py", line 14, in <module>
    from plugins.analyzers import *
  File "/usr/share/WiFi-Pumpkin/plugins/analyzers/image.py", line 5, in <module>
    from scapy_http import http
ImportError: No module named scapy_http

#### Please tell us details about your environment.
* Alfa AWSO36NHA 
* Version used tool: 0.8.5
* Virtual Machine (yes or no and which): no
* Operating System and version: 
Distributor ID:	Kali
Description:	Kali GNU/Linux Rolling
Release:	kali-rolling
Codename:	kali-rolling

Vadim555-555, 2017-12-30:
Hi ! sudo pip install scapy-http

root1825, 2017-12-30:
thanks solved

closed 2017-12-30 by root1825