P0cL4bs/WiFi-Pumpkin-deprecated #315: Failed building wheel for cryptography

Darkcast, 2017-12-31:
Installation failed 
 ----------------------------------------
  Can't rollback cryptography, nothing uninstalled.
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Tyh5Do/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-9vX9uj-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-Tyh5Do/cryptography/
----------------------------------------

Failed building wheel for cryptography
 Kali linux 2017.3
 Virtual Machine yes vmware  
* Operating System and version:  windows 10

yudevan, 2018-01-06:
Hi,
It happened to me after I did  dist-upgrade. apparently it caused by kali upgrading libssl to version 1.0.2-n1. 
I resolved this problem after downgrading libssl by:
apt-get install libssl1.0

and then I reinstall mitmproxy to version 0.18.2 (the last version with python2.7) the mitmproxy has to be the one with python2.7 as the recent release, mitmproxy has dropped support for python2.7. here is what i did 

pip install mitmproxy==0.18.2 typing

and you should be set

Darkcast, 2018-01-10:
Will do thanks

thecomicguy, 2018-01-15:
Yudeven, thank you for your succinct instructions. I am indeed - set. Worked perfectly.

cash87, 2018-01-15:
Hello guys.

I have that same issue but in Ubuntu 16.04. I did the downgrade libssl to 1.0 but still have the same answer

"Failed building wheel for cryptography"
Before that say me alot of message about methods DEPRECATEDIN_1_1_0

I test with ubuntu 14.04, in these work fine.

Thanks for your help

Regards

yudevan, 2018-01-16:
Could you paste the error here especially few lines before "Failed building wheel for crypto..."

cash87, 2018-01-16:
Sure, there is the info when i try to install

usage: ./installer.sh --install | --uninstall
Ign:1 http://dl.google.com/linux/chrome-remote-desktop/deb stable InRelease
Des:2 http://security.ubuntu.com/ubuntu xenial-security InRelease [102 kB]     
Obj:3 http://ppa.launchpad.net/noobslab/apps/ubuntu xenial InRelease           
Obj:4 http://mx.archive.ubuntu.com/ubuntu xenial InRelease                     
Ign:5 http://dl.google.com/linux/chrome/deb stable InRelease                   
Des:6 http://mx.archive.ubuntu.com/ubuntu xenial-updates InRelease [102 kB]    
Obj:7 https://download.sublimetext.com apt/stable/ InRelease                   
Obj:8 http://dl.google.com/linux/chrome-remote-desktop/deb stable Release      
Obj:9 http://ppa.launchpad.net/ondrej/php/ubuntu xenial InRelease              
Des:10 http://security.ubuntu.com/ubuntu xenial-security/main amd64 DEP-11 Metadata [62.4 kB]
Obj:12 http://dl.google.com/linux/chrome/deb stable Release                    
Obj:13 http://ppa.launchpad.net/ubuntu-wine/ppa/ubuntu xenial InRelease        
Des:15 http://security.ubuntu.com/ubuntu xenial-security/main DEP-11 64x64 Icons [57.6 kB]
Des:16 http://mx.archive.ubuntu.com/ubuntu xenial-backports InRelease [102 kB] 
Des:17 http://security.ubuntu.com/ubuntu xenial-security/universe amd64 DEP-11 Metadata [51.4 kB]
Des:18 http://security.ubuntu.com/ubuntu xenial-security/universe DEP-11 64x64 Icons [80.2 kB]
Obj:19 http://ppa.launchpad.net/webupd8team/sublime-text-3/ubuntu xenial InRelease
Des:20 http://mx.archive.ubuntu.com/ubuntu xenial-updates/main amd64 DEP-11 Metadata [307 kB]
Des:21 http://mx.archive.ubuntu.com/ubuntu xenial-updates/main DEP-11 64x64 Icons [222 kB]
Des:22 http://mx.archive.ubuntu.com/ubuntu xenial-updates/universe amd64 DEP-11 Metadata [185 kB]
Des:23 http://mx.archive.ubuntu.com/ubuntu xenial-updates/universe DEP-11 64x64 Icons [271 kB]
Des:24 http://mx.archive.ubuntu.com/ubuntu xenial-updates/multiverse amd64 DEP-11 Metadata [5 892 B]
Des:25 http://mx.archive.ubuntu.com/ubuntu xenial-backports/main amd64 DEP-11 Metadata [3 324 B]
Des:26 http://mx.archive.ubuntu.com/ubuntu xenial-backports/universe amd64 DEP-11 Metadata [4 716 B]
Descargados 1 558 kB en 2s (630 kB/s)                                   
Leyendo lista de paquetes... Hecho
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
libffi-dev ya está en su versión más reciente (3.2.1-4).
libxslt1-dev ya está en su versión más reciente (1.1.28-2.1ubuntu0.1).
zlib1g-dev ya está en su versión más reciente (1:1.2.8.dfsg-2ubuntu4.1).
python-pip ya está en su versión más reciente (8.1.1-2ubuntu0.4).
libssl-dev ya está en su versión más reciente (1.1.0g-2.1+ubuntu16.04.1+deb.sury.org+1).
libxml2-dev ya está en su versión más reciente (2.9.4-1+deb.sury.org~xenial+2).
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 20 no actualizados.
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
libarchive-dev ya está en su versión más reciente (3.1.2-11ubuntu0.16.04.3).
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 20 no actualizados.
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
build-essential ya está en su versión más reciente (12.1ubuntu2).
libnetfilter-queue-dev ya está en su versión más reciente (1.0.2-2).
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 20 no actualizados.
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
rfkill ya está en su versión más reciente (0.5-1ubuntu3).
python-qt4 ya está en su versión más reciente (4.11.4+dfsg-1build4).
python-scapy ya está en su versión más reciente (2.2.0-1).
hostapd ya está en su versión más reciente (1:2.4-0ubuntu6.2).
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 20 no actualizados.
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
python-dev ya está en su versión más reciente (2.7.11-1).
git ya está en su versión más reciente (1:2.7.4-0ubuntu1.3).
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 20 no actualizados.
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
libpcap-dev ya está en su versión más reciente (1.7.4-2).
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 20 no actualizados.
Requirement already satisfied: Twisted in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 1))
[... 1152 lines cut ...]
    /usr/include/openssl/x509.h:552:6: note: expected ‘const X509_ALGOR ** {aka const struct X509_algor_st **}’ but argument is of type ‘X509_ALGOR ** {aka struct X509_algor_st **}’
     void X509_get0_signature(const ASN1_BIT_STRING **psig,
          ^
    build/temp.linux-x86_64-2.7/_openssl.c: In function ‘_cffi_f_X509_get0_signature’:
    build/temp.linux-x86_64-2.7/_openssl.c:59559:25: warning: passing argument 1 of ‘X509_get0_signature’ from incompatible pointer type [-Wincompatible-pointer-types]
       { X509_get0_signature(x0, x1, x2); }
                             ^
    In file included from /usr/include/openssl/cms.h:16:0,
                     from build/temp.linux-x86_64-2.7/_openssl.c:542:
[... cut ...]
    ----------------------------------------
  Rolling back uninstall of cryptography
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-cJjek3/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-L3MF0d-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-cJjek3/cryptography/

closed 2018-01-20 by mh4x0f

Anucode, 2019-05-16:
I got the same error on windows 10. Can anyone help on this?.

Thank you.