P0cL4bs/WiFi-Pumpkin-deprecated #318: mitmproxy==0.18.2 Not Satisfied.

HITECHPR, 2018-01-08:
mitmproxy==0.18.2 Not Satisfied.
Installed is 2.02 (And is what KaliRolling has).

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter
* Version used tool: 0.8.5
* Virtual Machine (yes or no and which): Yes Virtual Box
* Operating System and version:  Kali 2017.3 Virtual Image
Thanks

yudevan, 2018-01-08:
hi
could you post the what command did you use, and also full debugging messages
I have the latest version of kali as well, and it is working fine after I did downgrade mitmproxy.
try pip2 install mitmproxy==0.18.2 
thx

HITECHPR, 2018-01-08:
[mitmproxy.log](https://github.com/P0cL4bs/WiFi-Pumpkin/files/1612889/mitmproxy.log)
pip2 install mitmproxy==0.18.2
 error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
   ----------------------------------------
  Can't rollback cryptography, nothing uninstalled.
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-6zc_jy/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-3rD6aH-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-6zc_jy/cryptography/

I have included the entire output in log file.

yudevan, 2018-01-09:
Which libssl version installed on your system? in order to make it work it has to be version 1.0. I had to downgrade mine by issuing apt-get install libssl1.0

HITECHPR, 2018-01-10:
That fixed it but the order is
apt-get install libssl1.0
pip2 install mitmproxy==0.18.2

Now I get to play....where's the documentation....especially the karma thing.
Thanks and keep up the good work.

yudevan, 2018-01-10:
of course,  Just like I mentioned on #315 ticket

HITECHPR, 2018-01-11:
So now I switched from a VM to a Raspberry PI3 (Image also downloaded from Kali).
Same Procedure,same error with extra.
Applied both libssl1.0 and pip2  mitmprox==0.18.2
Same error message about mitmprox
New error on startup
:0: UserWarning: You do not have a working installation of the service_identity module: 'cannot import name opentype'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.

I tried manually install the service_identity module and it state that it is already installed.
Thanks

VitaminPi, 2018-01-14:
I'm also on a Raspberry PI 3 and did 
apt-get install libssl1.0
pip2 install mitmproxy==0.18.2

I got the same error as HITECHPR after starting "wifi-pumpkin" 

:0: UserWarning: You do not have a working installation of the service_identity module: 'cannot import name opentype'. Please install it from https://pypi.python.org/pypi/service_identity and make sure all of its dependencies are satisfied. Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification. Many valid certificate/hostname mappings may be rejected.

By doing: 
pip install --upgrade service_identity

The error message goes away when starting pumpkin "wifi-pumpkin"

Before AND after doing the service_identity upgrade, the pumpkin application runs, but when I connect my device to the pumpkin's SSID, the device does not show up in the connected devices list (except the first time it connects) even though the device says it's connected and the pumpkin activity section shows activity when you try to go to a site on a browser. Also, the device won't connect to the Internet even though the PI can connect via its Ethernet connection to my router.

I noticed that upgrading service_identity upgrades the following components that installing mitmproxy 0.18.2 downgrades:

cryptography-2.1.4 idna-2.6 ipaddress-1.0.19 pyasn1-0.4.2 pyopenssl-17.5.0 six-1.11.0

Note: those are the upgraded component versions.

I'm not sure if the service_identity issue is related to issue I'm having. I'm wondering if HITECHPR tried upgrading or reinstalling service_identity and if so, does pumpkin then work without issues?

One more thing. Everytime I install mitmproxy, I get "Failed building wheel for Pillow." Not sure if that is significant. Got any ideas of what to try next?

root@kali# pip2 install mitmproxy==0.18.2
Collecting mitmproxy==0.18.2
  Using cached mitmproxy-0.18.2-py2.py3-none-any.whl
Requirement already satisfied: lxml<=3.6.0,>=3.5.0 in /usr/local/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Requirement already satisfied: cssutils<1.1,>=1.0.1 in /usr/local/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Requirement already satisfied: brotlipy<0.7,>=0.5.1 in /usr/local/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Collecting pyOpenSSL<17.0,>=16.0 (from mitmproxy==0.18.2)
  Using cached pyOpenSSL-16.2.0-py2.py3-none-any.whl
Requirement already satisfied: pyparsing<2.2,>=2.1.3 in /usr/local/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Collecting six<1.11,>=1.10 (from mitmproxy==0.18.2)
  Using cached six-1.10.0-py2.py3-none-any.whl
Requirement already satisfied: blinker<1.5,>=1.4 in /usr/local/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Requirement already satisfied: backports.ssl-match-hostname<3.6,>=3.5.0.1 in /usr/local/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Collecting cryptography<1.6,>=1.3 (from mitmproxy==0.18.2)
Requirement already satisfied: construct<2.6,>=2.5.2 in /usr/local/lib/python2.7/dist-packages/construct-2.5.2-py2.7.egg (from mitmproxy==0.18.2)
Requirement already satisfied: jsbeautifier<1.7,>=1.6.3 in /usr/local/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Requirement already satisfied: requests<2.12,>=2.9.1 in /usr/local/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Requirement already satisfied: tornado<4.5,>=4.3 in /usr/local/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Requirement already satisfied: pyperclip<1.6,>=1.5.22 in /usr/local/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Requirement already satisfied: certifi>=2015.11.20.1 in ./dist-packages (from mitmproxy==0.18.2)
Collecting Pillow<3.5,>=3.2 (from mitmproxy==0.18.2)
  Using cached Pillow-3.4.2.tar.gz
Collecting hyperframe<5,>=4.0.1 (from mitmproxy==0.18.2)
  Using cached hyperframe-4.0.2-py2.py3-none-any.whl
Collecting pyasn1<0.2,>=0.1.9 (from mitmproxy==0.18.2)
  Using cached pyasn1-0.1.9-py2.py3-none-any.whl
Collecting h2<2.5,>=2.4.1 (from mitmproxy==0.18.2)
  Using cached h2-2.4.4-py2.py3-none-any.whl
Collecting passlib<1.7,>=1.6.5 (from mitmproxy==0.18.2)
  Using cached passlib-1.6.5-py2.py3-none-any.whl
Requirement already satisfied: ipaddress<1.1,>=1.0.15; python_version == "2.7" in /usr/local/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Requirement already satisfied: Flask<0.12,>=0.10.1 in /usr/local/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Collecting configargparse<0.12,>=0.10 (from mitmproxy==0.18.2)
Requirement already satisfied: enum34<2,>=1.0.4; python_version == "2.7" in ./dist-packages (from mitmproxy==0.18.2)
Collecting typing==3.5.2.2; python_version == "2.7" (from mitmproxy==0.18.2)
Requirement already satisfied: click<7.0,>=6.2 in /usr/local/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Collecting watchdog<0.9,>=0.8.3 (from mitmproxy==0.18.2)
Collecting html2text<=2016.9.19,>=2016.1.8 (from mitmproxy==0.18.2)
Collecting urwid<1.4,>=1.3.1 (from mitmproxy==0.18.2)
Requirement already satisfied: cffi>=1.0.0 in /usr/local/lib/python2.7/dist-packages (from brotlipy<0.7,>=0.5.1->mitmproxy==0.18.2)
Requirement already satisfied: idna>=2.0 in /usr/local/lib/python2.7/dist-packages (from cryptography<1.6,>=1.3->mitmproxy==0.18.2)
Requirement already satisfied: setuptools>=11.3 in ./dist-packages (from cryptography<1.6,>=1.3->mitmproxy==0.18.2)
Requirement already satisfied: editorconfig>=0.12.0 in /usr/local/lib/python2.7/dist-packages (from jsbeautifier<1.7,>=1.6.3->mitmproxy==0.18.2)
[... cut ...]
    ValueError: jpeg is required unless explicitly disabled using --disable-jpeg, aborting

    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-AmhVTC/Pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-zClvZg-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-AmhVTC/Pillow/

yudevan, 2018-01-16:
the error you have is indicating that you are missing libjpeg-dev try to install that library.

VitaminPi, 2018-01-16:
I just ran:

sudo apt-get install libjpeg-dev

and successfully installed that library. This allowed me to build the Pillow wheel successfully when running:

pip2 install mitmproxy==0.18.2

Hooray!

Unfortunately, I still have the following issue: 

The pumpkin application runs, but when I connect my device to the pumpkin's SSID, the device does not show up in the connected devices list (except the first time it connects) even though the device says it's connected and the pumpkin activity section shows activity when you try to go to a site on a browser. Also, the device will not connect to the Internet even though the PI can connect via its Ethernet connection to my router. Here's the output from two attempts of starting, stopping, and connecting to the pumpkinAP.

root@kali:~# wifi-pumpkin
Loading GUI...
WiFi-Pumpkin Running!
 
[*] Loading debugging mode
[*] Current Session::ID [ODc2OTI=]
[*] Configuring hostapd...
[*] Enable forwarding in iptables...
[*] Configuring dhcpd...
[*] Sharing Internet Connections with NAT...
[New Thread 950 (hostapd)]
[New Thread 952 (Dns2Proxy)]
SSLstrip v0.9 + POC by Leonardo Nve
 
[*] TCPProxy running on port 80/8080:
 
TCPProxy::ftp               status:On
TCPProxy::hexdump           status:On
TCPProxy::kerberos          status:On
TCPProxy::summary           status:On
TCPProxy::imageCap          status:On
TCPProxy::httpCap           status:On
TCPProxy::emails            status:On
 
 
-------------------------------
AP::[pumpkinAP] Running...
AP::BSSID::[ac:e6:45:03:26:4d] CH 11
WARNING:root:POST Data (update.googleapis.com):
<?xml version="1.0" encoding="UTF-8"?><request protocol="3.1" dedup="cr" acceptformat="crx2,crx3" version="chrome-63.0.3239.73" prodversion="63.0.3239.73" requestid="{77f44b4d-3c27-4b9c-bd27-1300d0ce8f4c}" lang="en-US" updaterchannel="" prodchannel="" os="mac" arch="arm64" nacl_arch="arm64"><hw physmemory="2"/><os platform="iOS" arch="iPhone9,3" version="11.2.2"/><app appid="hfnkpimlhhgieaddgfemjhofmfblmnib" version="4218" brand="GOYI" cohort="1:jcl:" cohortname="Auto" enabled="1"><updatecheck/><ping rd="4031" ping_freshness="{5f637244-9455-471a-b2af-9c3f513061b8}"/></app></request>
-------------------------------
Thread::[hostapd] successfully stopped.
Thread::[Dns2Proxy] successfully stopped.
Thread::[sslstrip2] successfully stopped.
Thread::[TCPProxy] successfully stopped.
 
[*] Loading debugging mode
[*] Current Session::ID [Njg0OTM=]
[*] Configuring hostapd...
[*] Enable forwarding in iptables...
[*] Configuring dhcpd...
[*] Sharing Internet Connections with NAT...
[New Thread 1033 (hostapd)]
[New Thread 1035 (Dns2Proxy)]
SSLstrip v0.9 + POC by Leonardo Nve
 
[*] TCPProxy running on port 80/8080:
 
TCPProxy::ftp               status:On
TCPProxy::hexdump           status:On
TCPProxy::kerberos          status:On
TCPProxy::summary           status:On
TCPProxy::imageCap          status:On
TCPProxy::httpCap           status:On
TCPProxy::emails            status:On
 
[... 32 lines cut ...]
    self.addNextWidget(self.APclients[agent])
  File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 51, in addNextWidget
    self.items_widgets[agent.keys()[0]] = {}
AttributeError: 'str' object has no attribute 'keys'
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 1064, in get_Hostapd_Response
    self.TabInfoAP.delete_item(data)
  File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 71, in delete_item
    self.addNextWidget(self.APclients[agent])
  File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 51, in addNextWidget
    self.items_widgets[agent.keys()[0]] = {}
AttributeError: 'str' object has no attribute 'keys'
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 1064, in get_Hostapd_Response
    self.TabInfoAP.delete_item(data)
  File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 71, in delete_item
    self.addNextWidget(self.APclients[agent])
  File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 51, in addNextWidget
    self.items_widgets[agent.keys()[0]] = {}
AttributeError: 'str' object has no attribute 'keys'
-------------------------------
Thread::[hostapd] successfully stopped.
Thread::[Dns2Proxy] successfully stopped.
Thread::[sslstrip2] successfully stopped.
Thread::[TCPProxy] successfully stopped.

yudevan, 2018-01-16:
this problem I have to dig deeper. seems it caused by wrong type identification. for  sure @mh4x0f  can help with this

mh4x0f, 2018-01-19:
@VitaminPi create an new issue for this bug.  @yudevan thanks for support :D

VitaminPi, 2018-01-19:
New issue created

[#322](https://github.com/P0cL4bs/WiFi-Pumpkin/issues/322)

closed 2018-01-20 by mh4x0f