P0cL4bs/WiFi-Pumpkin-deprecated #319: wifi-pumpkin

Viveksgautam, 2018-01-12:
Collecting netfilterqueue
  Using cached NetfilterQueue-0.8.1.tar.gz
Building wheels for collected packages: netfilterqueue
  Running setup.py bdist_wheel for netfilterqueue ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-VV1f3e/netfilterqueue/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp0vswKjpip-wheel- --python-tag cp27:
  running bdist_wheel
  running build
  running build_ext
  building 'netfilterqueue' extension
  creating build
  creating build/temp.linux-x86_64-2.7
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-HVkOs2/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c netfilterqueue.c -o build/temp.linux-x86_64-2.7/netfilterqueue.o
  netfilterqueue.c:437:42: fatal error: libnfnetlink/linux_nfnetlink.h: No such file or directory
   #include "libnfnetlink/linux_nfnetlink.h"
                                            ^
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for netfilterqueue
  Running setup.py clean for netfilterqueue
Failed to build netfilterqueue
Installing collected packages: netfilterqueue
  Running setup.py install for netfilterqueue ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-VV1f3e/netfilterqueue/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-OWfDiJ-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_ext
    building 'netfilterqueue' extension
    creating build
    creating build/temp.linux-x86_64-2.7
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-HVkOs2/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c netfilterqueue.c -o build/temp.linux-x86_64-2.7/netfilterqueue.o
    netfilterqueue.c:437:42: fatal error: libnfnetlink/linux_nfnetlink.h: No such file or directory
     #include "libnfnetlink/linux_nfnetlink.h"
                                              ^
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-VV1f3e/netfilterqueue/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-OWfDiJ-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-VV1f3e/netfilterqueue/## What's the problem (or question)?

*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: 
* Virtual Machine (yes or no and which): 
* Operating System and version:

yudevan, 2018-01-12:
Hi
You will need to install libnfnetlink-dev on your linux distribution. 
I think i also installed python-nfqueue along with that

Viveksgautam, 2018-01-17:
# pip install NetfilterQueue
Collecting NetfilterQueue
  Using cached NetfilterQueue-0.8.1.tar.gz
Building wheels for collected packages: NetfilterQueue
  Running setup.py bdist_wheel for NetfilterQueue ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-qtEKUP/NetfilterQueue/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpKH8bkVpip-wheel- --python-tag cp27:
  running bdist_wheel
  running build
  running build_ext
  building 'netfilterqueue' extension
  creating build
  creating build/temp.linux-x86_64-2.7
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-VlMpWk/python2.7-2.7.14=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c netfilterqueue.c -o build/temp.linux-x86_64-2.7/netfilterqueue.o
  netfilterqueue.c:437:10: fatal error: libnfnetlink/linux_nfnetlink.h: No such file or directory
   #include "libnfnetlink/linux_nfnetlink.h"
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for NetfilterQueue
  Running setup.py clean for NetfilterQueue
Failed to build NetfilterQueue
Installing collected packages: NetfilterQueue
  Running setup.py install for NetfilterQueue ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-qtEKUP/NetfilterQueue/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-R4NTF3-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_ext
    building 'netfilterqueue' extension
    creating build
    creating build/temp.linux-x86_64-2.7
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-VlMpWk/python2.7-2.7.14=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c netfilterqueue.c -o build/temp.linux-x86_64-2.7/netfilterqueue.o
    netfilterqueue.c:437:10: fatal error: libnfnetlink/linux_nfnetlink.h: No such file or directory
     #include "libnfnetlink/linux_nfnetlink.h"
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-qtEKUP/NetfilterQueue/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-R4NTF3-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-qtEKUP/NetfilterQueue/

yudevan, 2018-01-19:
which distribution do you use? for example in kali, you will need to do apt-get install libnfnetlink-dev

closed 2018-01-20 by mh4x0f

Viveksgautam, 2018-01-21:
i am use kali 64 bit 2017.3 i am also tryed apt-get install libnfnetlink-dev