P0cL4bs/WiFi-Pumpkin-deprecated #321: mitmproxy==0.18.2 Installed but, the Gui says "Not Satisfied"

gitxmax, 2018-01-15:
Hello, 
I did s fresh install of Pumpkin and it went smooth, to the very end, installer showed no errors.
But, when I go to Pumpkin-Proxy tab of the gui, it says  "the package requirement mitmproxy==0.18.2 is not satisfied"
I tried reinstalling mitmproxy==0.18.2 with pip and pip2, but it had no effect.
Running on PI B3+ with the latest Kali, if that helps.

yudevan, 2018-01-16:
can you try to execute mitmproxy from command shell, if your mitmproxy is okay then you will be displayed clean window with blue bar on the bottom, otherwise, please paste the error log

gitxmax, 2018-01-18:
Thanks, that helped, the issue was with the UTF-8 encoding. Can close the ticket.

closed 2018-01-20 by mh4x0f

rentandleave, 2018-08-16:
Just reopening this . Since i Still have the problem with :mitmproxy==0.18.2 Not Satisfied " on the WiFi-Pumpkin GUI. . How did the issue with UTF-8 encoding solved the problem.It will be very helpful.

Thanks in Advance