P0cL4bs/WiFi-Pumpkin-deprecated #322: Device not showing in Pumpkin device list. No connection to Internet.

VitaminPi, 2018-01-19:
#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: 
* Virtual Machine (yes or no and which): 
* Operating System and version: 

## What's the problem (or question)?

I've got Kali 2017.3 on a Raspberry Pi 3
I've ran the following:

apt-get install libssl1.0
apt-get install libjpeg-dev
pip2 install mitmproxy==0.18.2
pip install --upgrade service_identity

I have the following issue:

The pumpkin application runs, but when I connect my device to the pumpkin's SSID, the device does not show up in the connected devices list (except the very first time the device connects) even though the device says it's connected and the pumpkin activity section shows activity when you try to go to a site on a browser. Also, the device will not connect to the Internet even though the PI can connect via its Ethernet connection to my router. Here's the output from two attempts of starting, stopping, and connecting to the pumpkinAP.

root@kali:~# wifi-pumpkin
Loading GUI...
WiFi-Pumpkin Running!

[] Loading debugging mode
[] Current Session::ID [ODc2OTI=]
[] Configuring hostapd...
[] Enable forwarding in iptables...
[] Configuring dhcpd...
[] Sharing Internet Connections with NAT...
[New Thread 950 (hostapd)]
[New Thread 952 (Dns2Proxy)]
SSLstrip v0.9 + POC by Leonardo Nve

[*] TCPProxy running on port 80/8080:

TCPProxy::ftp status:On
TCPProxy::hexdump status:On
TCPProxy::kerberos status:On
TCPProxy::summary status:On
TCPProxy::imageCap status:On
TCPProxy::httpCap status:On
TCPProxy::emails status:On

AP::[pumpkinAP] Running...
AP::BSSID::[ac:e6:45:03:26:4d] CH 11
WARNING:root:POST Data (update.googleapis.com):

Thread::[hostapd] successfully stopped.
Thread::[Dns2Proxy] successfully stopped.
Thread::[sslstrip2] successfully stopped.
Thread::[TCPProxy] successfully stopped.

[] Loading debugging mode
[] Current Session::ID [Njg0OTM=]
[] Configuring hostapd...
[] Enable forwarding in iptables...
[] Configuring dhcpd...
[] Sharing Internet Connections with NAT...
[New Thread 1033 (hostapd)]
[New Thread 1035 (Dns2Proxy)]
SSLstrip v0.9 + POC by Leonardo Nve

[*] TCPProxy running on port 80/8080:

TCPProxy::ftp status:On
TCPProxy::hexdump status:On
TCPProxy::kerberos status:On
TCPProxy::summary status:On
TCPProxy::imageCap status:On
TCPProxy::httpCap status:On
TCPProxy::emails status:On

AP::[pumpkinAP] Running...
AP::BSSID::[ac:e6:45:03:26:4d] CH 11
Traceback (most recent call last):
File "/usr/share/WiFi-Pumpkin/core/main.py", line 1064, in get_Hostapd_Response
self.TabInfoAP.delete_item(data)
File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 71, in delete_item
self.addNextWidget(self.APclients[agent])
File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 54, in addNextWidget
for client in agent[key].keys():
AttributeError: 'str' object has no attribute 'keys'
Traceback (most recent call last):
File "/usr/share/WiFi-Pumpkin/core/main.py", line 1064, in get_Hostapd_Response
self.TabInfoAP.delete_item(data)
File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 71, in delete_item
self.addNextWidget(self.APclients[agent])
File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 51, in addNextWidget
self.items_widgets[agent.keys()[0]] = {}
AttributeError: 'str' object has no attribute 'keys'
Traceback (most recent call last):
File "/usr/share/WiFi-Pumpkin/core/main.py", line 1064, in get_Hostapd_Response
self.TabInfoAP.delete_item(data)
File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 71, in delete_item
self.addNextWidget(self.APclients[agent])
File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 51, in addNextWidget
self.items_widgets[agent.keys()[0]] = {}
AttributeError: 'str' object has no attribute 'keys'
Traceback (most recent call last):
File "/usr/share/WiFi-Pumpkin/core/main.py", line 1064, in get_Hostapd_Response
self.TabInfoAP.delete_item(data)
File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 71, in delete_item
self.addNextWidget(self.APclients[agent])
File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 51, in addNextWidget
self.items_widgets[agent.keys()[0]] = {}
AttributeError: 'str' object has no attribute 'keys'
Traceback (most recent call last):
File "/usr/share/WiFi-Pumpkin/core/main.py", line 1064, in get_Hostapd_Response
self.TabInfoAP.delete_item(data)
File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 71, in delete_item
self.addNextWidget(self.APclients[agent])
File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 51, in addNextWidget
self.items_widgets[agent.keys()[0]] = {}
AttributeError: 'str' object has no attribute 'keys'
Traceback (most recent call last):
File "/usr/share/WiFi-Pumpkin/core/main.py", line 1064, in get_Hostapd_Response
self.TabInfoAP.delete_item(data)
File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 71, in delete_item
self.addNextWidget(self.APclients[agent])
File "/usr/share/WiFi-Pumpkin/core/widgets/customiseds.py", line 51, in addNextWidget
self.items_widgets[agent.keys()[0]] = {}
AttributeError: 'str' object has no attribute 'keys'
Thread::[hostapd] successfully stopped.
Thread::[Dns2Proxy] successfully stopped.
Thread::[sslstrip2] successfully stopped.
Thread::[TCPProxy] successfully stopped.

mh4x0f, 2018-01-20:
I don't find any problem with this function for remove disconnected client.

mh4x0f, 2018-01-20:
try start WP and send me the output of command  bellow:
``` sh 
sudo iptables -t nat -L
```
Obs: with this ouput, I'll check if alright with internet connection.

VitaminPi, 2018-01-21:
root@kali:~# sudo iptables -t nat -L
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination

Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination

HSN80, 2018-02-19:
I have this problem too

xXD3F4LTXx, 2018-05-03:
I had this issue. Switching from a Class-C address to a Class-A address fixed it for me

2fst4u, 2018-10-25:
I'm getting this issue too. I have tried class A, B and C, and I have tried changing the dhcp server and I still see no devices

closed 2018-10-28 by mh4x0f

hurton2002, 2019-12-08:
mee tooo

hurton2002, 2019-12-08:
i cant see devices

commit 096eb159 (2018-10-28): fixed #322 error when remove the client from tableWidget
commit f716f670 (2018-10-28): fixed #322 error when remove the client from tableWidget