P0cL4bs/WiFi-Pumpkin-deprecated #323: ImportError: cannot import name ssl_match_hostname / mitmproxy==0.18.2

gitxmax, 2018-01-20:
Totally got stuck with getting mitmproxy==0.18.2 to run on RPI3 with the latest Kali 17.3
The install goes smooth, not a single error displayed.
But, when I try to start up mitmproxy, I get the following error:
"from backports import ssl_match_hostname
ImportError: cannot import name ssl_match_hostname"

Surely, I tried reinstalling blackports.ssl_match_hostname many times over, followed by pip install -U mitmproxy==0.18.2

Also, tried clean install, with slightly different approaches, like 6 or 7 times, but, always ends up with this error.

gitxmax, 2018-01-23:
Alright, I failed to ever resolve the issue on Kali 17.3, posted the error on mitmproxy git, but they simply closed the issue, with the comment, to upgrade mitmproxy to the latest version. 

So, I flashed the Kali 17.1 Today and installed Wifi-Pumpkin again.
The only issue I had, is that service_identity module wasn't being recognized, despite being installed wit pip.
The solution was to use easy_install instead, which complained about an outdated dependency.
Once the dependency was updated, reinstalled the service_identity and WiFi-Pumpkin stated up with no mistakes.

Apart from the Wifi-Pumpkin, I was having other multiple issues, running Kali 17.3
So, for now, I recommend to use the earlier version of Kali, on Raspberry.
can close the issue now...

ayushg162, 2018-02-12:
Import Error: No module named scapy http

closed 2018-03-04 by mh4x0f

FrankBWalsh, 2018-03-09:
I'm experiencing this same issue where Wifi-Pumpkin is complaining about out of date service_identity module when I start it. Specifically the message reads:

You do not have a working installation of the service_identity module: 'cannot import name opentype' Please install it from <https://pypi.python.org/pypi/service_idenity and make sure all of its dependecies are satisfied. Without the service_identity module, Twisted can preform only rudimentary TLS client hostname verification. Many valid certificate/hostname mappings may be rejected.

I've tried doing easy_install of service_identity and that doesn't seem to solv anything. I am also using the latest Kali.

How do I resolve this?

HynekPetrak, 2019-06-15:
It seems the pip installs the backports package into a /usr/local/lib while other backports packages installed on Kali with dpkg reside in /usr/lib

This resolved for me on 2019.2:
```
pip2 uninstall backports.ssl_match_hostname

apt install python-backports.ssl-match-hostname
```