P0cL4bs/WiFi-Pumpkin-deprecated #326: struct.error: argument for 's' must be a string <Strikes Again>

gitxmax, 2018-01-24:
Hello,
I have a fresh install of Pumpkin.

When I launch the "ARP POISONER" module, i get this error:

Traceback (most recent call last): File "/usr/share/WiFi-Pumpkin/core/main.py", line 680, in show_arp_posion self.Farp_posion = GUIModules.frm_Arp_Poison(self.window_phishing) File "/usr/share/WiFi-Pumpkin/modules/poisoners/arp_poisoner.py", line 42, in __init__ self.GUI() File "/usr/share/WiFi-Pumpkin/modules/poisoners/arp_poisoner.py", line 145, in GUI self.ConfigureEdits() File "/usr/share/WiFi-Pumpkin/modules/poisoners/arp_poisoner.py", line 174, in ConfigureEdits self.txt_mac.setText(Refactor.getHwAddr(x['activated'][0])) File "/usr/share/WiFi-Pumpkin/core/utils.py", line 237, in getHwAddr info = ioctl(s.fileno(), 0x8927, pack('256s', ifname[:15])) struct.error: argument for 's' must be a string

I tried the solution offered here to modify it to: return get_if_addr("{0}".format(card))

This had no effect, same error persist.

Running Kali 17.1 on raspberry 3

keviiin38, 2018-03-13:
I had the same problem, i think that your AP isn't started ?
Tell me if I'm wrong
I solved this error by trying when the AP was started

mh4x0f, 2018-03-13:
@keviiin38 the arp poisoner it not necessary when AP started, I'll lock this option coming soon.

closed 2018-05-01 by mh4x0f