P0cL4bs/WiFi-Pumpkin-deprecated #329: Wifi card is broken after using tool

tigerwoods1, 2018-01-25:
## What's the problem (or question)?
After running the tool for the first time and then exiting; the laptops WiFi interface becomes unusable. My WiFi indicates that there is no WiFi connections around. THis makes it so I can't run the tool again with WiFi because i have only one AP(external card). PLEASE HELP! 

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): Card on multiple labtops
* Version used tool: newest
* Virtual Machine (yes or no and which):  Hell No
* Operating System and version: kali 2017.3

almaefogo, 2018-02-05:
Hello i have the same problem did you managed to fix it.

argum77, 2018-02-05:
No i havnt been able too

almaefogo, 2018-02-05:
For the moment i have given up about the kali in dual boot, currently downloading kali linux for the virtual box and i will try to use brigde mode for the internet connection and the usb adapter for the AP

Darzl, 2018-02-09:
'sudo service network-manager restart' worked for me

frostKn, 2018-02-23:
even after restarting my network manager my wifi card is still down.....help

closed 2018-03-04 by mh4x0f