P0cL4bs/WiFi-Pumpkin-deprecated #330: Error on Hostapd

Giova84, 2018-01-28:
## What's the problem (or question)?

I setup an AP (on wlan0), and as soon I clicked on start, an alert window appeared by telling me that in the hostapd.conf file there were errors: invalid AP name and invalid MAC Address.
I opened such file: /core/config/hostapd/hostapd+.conf;  but on line 4 and 5 these entries were not present at all. Seems not an error of my configuration (or at least not completely) since other tools which made AP by using Hostapd (eg fluxion and wifiphisher) are able to make the same AP, with the same entries (name and MAC address).

#### Details about my environment:
* Card wireless adapters name (please check if support AP/mode):
TP-LINK TL-WN722N; supports AP mode.

* Version used tool: 
0.8.5

* Virtual Machine (yes or no and which): 
Yes: VMware Workstation Player 12 on Windows 7 ultimate 64 bit

* Operating System and version: 
Kali Linux kali-linux-2018-W04-amd64
Linux h725 4.14.0-kali3-amd64 #1 SMP Debian 4.14.13-1kali1 (2018-01-25) x86_64 GNU/Linux

Screenshot attacched:

[image]

Giova84, 2018-01-28:
I forgot to say: as I added ssid and bssid, manually, to the config file, th AP is flawlessly started.

frostKn, 2018-02-23:
can u tell me how u did that

closed 2018-03-04 by mh4x0f