P0cL4bs/WiFi-Pumpkin-deprecated #331: Question about usage

Giova84, 2018-01-28:
## What's the problem (or question)? 
I have  a question, and I made some research before to ask.
I'm trying to figure out how to redirect tcp traffic to a specific IP adress (another website, but would be doable also with an html redirect, so by using an ad-hoc tag inside a php or html page);
Seems that I have to open the DNS Spoofer and Phishing Manager, and start the redirect attack. But seems that not work. Or I miss something? Maybe the redirect attack is availabe via plugins? All the plugins that I see are grayed out.

Details about my environment:

    Card wireless adapters name (please check if support AP/mode):
    TP-LINK TL-WN722N; supports AP mode.

    Version used tool:
    0.8.5

    Virtual Machine (yes or no and which):
    Yes: VMware Workstation Player 12 on Windows 7 ultimate 64 bit

    Operating System and version:
    Kali Linux kali-linux-2018-W04-amd64
    Linux h725 4.14.0-kali3-amd64 #1 SMP Debian 4.14.13-1kali1 (2018-01-25) x86_64 GNU

Giova84, 2018-01-28:
I don't know if is related, but when I set something in SSLStrip+Proxy, I  get:
`AttributeError: 'WifiPumpkin' object has no attribute 'set_proxy_scripts'`
I need to install something with pip?

Darzl, 2018-02-09:
I have done this setup https://xeushack.com/phishing-with-a-rogue-wifi-access-point/ (bottom of page) but still no redirect, when connected to access point I can ip direct to the page.

closed 2018-03-04 by mh4x0f

Fryles, 2019-06-06:
Currently having this problem as well on raspberry Pi 3 b+. Everything else works but when I use any functionality of the sslstrip+proxy plugin I get that exact attribute error.