P0cL4bs/WiFi-Pumpkin-deprecated #335: my wlan0 stopped suddenly and wlan1 is only working

ronak9099, 2018-02-23:
## What's the problem (or question)?

I was trying to use wifipumpkin and suddenly my wlan0 stopped and wlan1 is working..even when i remove the wlan1 my wlan0 doesnt search for any network..though i can see it in my iwconfig command. plz help

*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: 
* Virtual Machine (yes or no and which): 
* Operating System and version:

yudevan, 2018-02-23:
Which interface connected to internet initially?its because WP is doing rfkill to the interface which you are going to use as AP.

To solve it just try to restart network-manager service

frostKn, 2018-02-23:
restarting network manager dosent works

ronak9099, 2018-02-23:
my wlan0 was connected initiatlly and i inserted my usb adaptor and tried wifi pumpkin and it was not getting started after clicking on start..and something happen got some error and my wlan0 stopped now only usb wifi works..but i can see it working on win10 with no issues - i hav dual booted. Also, iwconfig shows my wifi, i tried airmon-ng it works and searches the networks..i think its just disabled somewhere but dont know where

frostKn, 2018-02-26:
use the command "rfkill" to check the status if everything is enabled then use "service network-manager restart" see if that fixes it for u.

closed 2018-03-04 by mh4x0f