P0cL4bs/WiFi-Pumpkin-deprecated #341: captive portal with wifi-pumpkin

Deniero, 2018-03-02:
## What's the problem (or question)?
I know this is quite a large question (or multiple) but I would really appreciate if somebody could take the time and help me with them:
I want to make a wireless access point with wifi-pumpkin, that, in the beginning, redirects the user to a startup page (captive portal) which is running on a local server. The captive portal forces the user to enter their email address and to accept the terms of use.
I found two good tutorials on the web 
(https://projectzme.wordpress.com/2012/03/19/captive-portal-using-php-and-iptables-firewall-on-linux/ and 
http://www.andybev.com/index.php/Using_iptables_and_PHP_to_create_a_captive_portal (both are very simmilar)) on how to do this but I ran into a bunch of problems.
The idea is fairly simple: i want to redirect every unknown users http traffic to my local server.
when the user agrees to my terms of use and entered his email, his traffic won't be redirected anymore...

This is what I have accomplished so far:
I started a local webserver  with the 'phishing manager' from wifi-pumpkin
than i added following iptables to wifi-pumpkin:

- `iptables -t mangle -N internet`

- `iptables -t mangle -A PREROUTING -i wlan0 -p tcp -m tcp --dport 80 -j internet`

- `iptables -t mangle -A internet -j MARK --set-mark 99`

- `iptables -t nat -A PREROUTING -i wlan0 -p tcp -m mark --mark 99 -m tcp --dport 80 -j DNAT --to-destination 192.168.3.1`

then i started the ap with wifi-pumpkin
now, every user gets redirected to the my captive portal (i tested it and it works).

So... here are (finally) my questions:

1. How is the correlation between the 'normal' iptables of my operating system and those of wifi-pumpkin? I don't know if this is a silly question but i am asking because when i edit those of my os, the iptables from wifi-pumpkin don't change, so i need to change them in the wifi-pumpkin settings.

2. Every user that that has submitted their email and agreed to the terms of use, should not be redirected to my local server. This means that I somehow need to remove  _mark 99_  (which everyone gets when connecting to my ap (because of my iptables from above)) for this user.
The tutorial makes this with the following comands (after they get the mac address of the user)
`sudo iptables -D internet -t mangle -m mac --mac-source USER_MAC_ADDRESS -j RETURN`
`sudo rmtrack USER_IP_ADDRESS`
I don't really know how to implement to capturing of the mac-address and also how to implement the two commands from above in wifi-pumpkin.

I would really appreciate if somebody could take the time and help me with my problem;)

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): CSL 300Mbit Wlan Adatpter (supports AP-mode)
* Version used tool: v0.8.5
* Virtual Machine (yes or no and which): Kali Linux v2018.1 running on vmware
* Operating System and version: Linux Mint v18.3

mh4x0f, 2018-03-12:
hi, nice report..

> How is the correlation between the 'normal' iptables of my operating system and those of wifi-pumpkin? I don't know if this is a silly question but i am asking because when i edit those of my os, the iptables from wifi-pumpkin don't change, so i need to change them in the wifi-pumpkin settings.

the WP use the iptables installed from OS, the tab iptables on WP only show you the commands that will be executed when started the access point.

> Every user that that has submitted their email and agreed to the terms of use, should not be redirected to my local server. This means that I somehow need to remove mark 99 (which everyone gets when connecting to my ap (because of my iptables from above)) for this user.
The tutorial makes this with the following comands (after they get the mac address of the user)
sudo iptables -D internet -t mangle -m mac --mac-source USER_MAC_ADDRESS -j RETURN
sudo rmtrack USER_IP_ADDRESS
I don't really know how to implement to capturing of the mac-address and also how to implement the two commands from above in wifi-pumpkin.

this is a feature that will be implemented on 0.8.6 version. anywhere , you can implement the feature using the server in php/python whatever. look this project [captive_portal](https://github.com/AloysAugustin/captive_portal)

Deniero, 2018-03-14:
Thanks for your answer;)

necrose99, 2018-09-05:
https://github.com/audibleblink/meterpreter , 
http://www.hackingarticles.in/embedded-backdoor-image-using-fakeimageexploiter/

(https://github.com/Tenticles-PTF/Evilgate  havin to modernize.... 
this  then having too add wifi pumpkin bridges ... 
IoT pumpkin to pc vpn bride (Manager) ( IoT sniffer ) 

https://github.com/P0cL4bs/WiFi-Pumpkin/issues/253
https://github.com/P0cL4bs/WiFi-Pumpkin/issues/351

I'd be tickled pink to have the Rouge WIFI Pumpkin captive fake portal module
5-G Cellular wireless coming soon , 1-2 GIGS uplink... just enough to feed saps Facebook itch ... and thank you for helping the redteam out... saps... 
(a thanks for using wifi https://github.com/trustedsec/social-engineer-toolkit just incase... ) 

https://www.dropbox.com/home/sakaki--gentoo-on-rpi3-64bit/packages-rpi3-arm64 
(https://www.dropbox.com/sh/4u9gbhps0ek402e/AAC5VLBE32cwH2gGjkj0Yz9sa?dl=0)

for now can take look at https://github.com/AloysAugustin/captive_portal as it might be easier to subvert to those ends.... than the tollgate... 

With Rock64 or RPI3 in 64bit mode very easy to hide and get my shells inside..   "Free Wall of Sheep wifi anyhone"  most will let their guard drop in the Cafe on corp campus..

mh4x0f, 2019-04-06:
@Deniero it out ! ;) thanks for suggestion

closed 2019-04-07 by mh4x0f