P0cL4bs/WiFi-Pumpkin-deprecated #344: Devices can't connect to AP

holao09, 2018-03-05:
Devices can't connect to AP

I setup all things seem OK. 
But, no devices can connect AP.
My cellphone say wrong password although no security password on AP.
My notebook can connect but no ip received and it's disconnected after.
I'll put the pictures later.

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: **0.8.5**
* Virtual Machine (yes or no and which): No
* Operating System and version: Kali Rolling 2018.1

closed 2018-03-12 by mh4x0f

poxyran, 2018-05-14:
I have the same problem. I have an up to date Pi3, WP starts without any apparent problem on the terminal but I cannot connect with my Android device or PC (Windows).

remixdance, 2018-08-17:
same problem here #344