P0cL4bs/WiFi-Pumpkin-deprecated #346: You do not have a working installation of the service_identity module: 'cannot import name opentype'

FrankBWalsh, 2018-03-09:
I'm experiencing this same issue where Wifi-Pumpkin is complaining about out of date service_identity module when I start it. Specifically the message reads:

You do not have a working installation of the service_identity module: 'cannot import name opentype' Please install it from <https://pypi.python.org/pypi/service_idenity and make sure all of its dependecies are satisfied. Without the service_identity module, Twisted can preform only rudimentary TLS client hostname verification. Many valid certificate/hostname mappings may be rejected.

I've tried doing easy_install of service_identity and that doesn't seem to solv anything. I am also using the latest Kali.

How do I resolve this?

closed 2018-03-09 by mh4x0f

f3max, 2018-10-22:
pip install service_identity
pip install --upgrade pyasn1-modules

mother2110, 2018-10-30:
[root:~]# wifi-pumpkin 
WiFi-Pumpkin need PyQt4 :(
[root:~]# 

Tried everything. Using Kali 2018.4

fir0j, 2018-12-16:
[-] Compiling binary hostapd-mana
fatal: No annotated tags can describe 'd98fe2724f2044773cea698beefdbb61313ab390'.
However, there were unannotated tags: try --tags.
/usr/bin/ld: cannot find -lnl-genl-3
collect2: error: ld returned 1 exit status
make: *** [Makefile:1064: hostapd] Error 1
[+] Install binary hostapd-mana...
cp: cannot stat 'hostapd': No such file or directory
cp: cannot stat 'hostapd_cli': No such file or directory
[+] hostapd-mana successful installation...

kanken57, 2019-01-25:
Had the same problem in Kali 2018.4 rolling and the "pip install --upgrade pyasn1-modules" is the only thing that worked after trying everything else here and else where. Thank you.

dea4ever, 2019-04-22:
> Had the same problem in Kali 2018.4 rolling and the "pip install --upgrade pyasn1-modules" is the only thing that worked after trying everything else here and else where. Thank you.

It work! tnx!

hubert10, 2019-05-05:
Hello, Just need to upgrade your service_identity by running the following command:

 pip3 install service_identity --force --upgrade

This works for me!

mTeden, 2019-05-24:
hey - on parrotos 4.19 i can't get service_identity to work properly. every dependency is fine. is there an workaround for parrot or something?

kaySpaz, 2019-06-23:
> pip install service_identity
> pip install --upgrade pyasn1-modules

Thanks allot! I forgot to update them. Jogged my memory

hiddenwishes, 2020-01-15:
still no luck with all solutions

> :0: UserWarning: You do not have a working installation of the service_identity module: 'cannot import name opentype'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
> /usr/local/lib/python2.7/dist-packages/bs4/element.py:16: UserWarning: The soupsieve package is not installed. CSS selectors cannot be used.
>   'The soupsieve package is not installed. CSS selectors cannot be used.'
> wifi-pumpkin.py: cannot connect to X server

when i use 

> ```
> pip install service_identity
> pip install --upgrade pyasn1-modules

 pip install --upgrade pyasn1-modules
Requirement already up-to-date: pyasn1-modules in /usr/local/lib/python2.7/dist-packages (0.2.8)
Collecting pyasn1<0.5.0,>=0.4.6 (from pyasn1-modules)
  Using cached https://files.pythonhosted.org/packages/62/1e/a94a8d635fa3ce4cfc7f506003548d0a2447ae76fd5ca53932970fe3053f/pyasn1-0.4.8-py2.py3-none-any.whl
mitmproxy 0.18.2 has requirement pyasn1<0.2,>=0.1.9, but you'll have pyasn1 0.4.8 which is incompatible.
Installing collected packages: pyasn1
Successfully installed pyasn1-0.4.8

```

jouellnyc, 2020-01-28:
> still no luck with all solutions
> > pip install service_identity
> > pip install --upgrade pyasn1-modules

Ditto. FWIW:

 ` $ uname -a
Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
 $ which python
/usr/bin/python
 $ /usr/bin/python -V
Python 2.7.13
`