P0cL4bs/WiFi-Pumpkin-deprecated #349: the device can use internet but can't open any website

MrBeanc, 2018-03-21:
## What's the problem (or question)?
i use the wifi-pumpkin with no proxy server like this
[image]
my device's app can use internet normally,but i can't open any website.My device can have right ip,right DNS and right gateway.If I use SSL+dns2proxy，i can open the website.
[image]
i try different class-address,but it still did not work.
the home also don't have the device which linked.

[image]

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
 my wireless adapters is TL-WN722N
 Version used tool: 
* Virtual Machine (yes or no and which): 
i use vmware
* Operating System and version: 
my operating system is kali2.0

my iptables:
root@kali:~# sudo iptables -t nat -L && ifconfig
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         

Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         
MASQUERADE  all  --  anywhere             anywhere            
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.157  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::20c:29ff:fe44:e301  prefixlen 64  scopeid 0x20<link>
        ether 00:0c:29:44:e3:01  txqueuelen 1000  (Ethernet)
        RX packets 8925  bytes 4801423 (4.5 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 8898  bytes 4373808 (4.1 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 24  bytes 1356 (1.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 24  bytes 1356 (1.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1400
        inet 172.16.0.1  netmask 255.240.0.0  broadcast 172.31.255.255
        inet6 fe80::bef6:85ff:fe03:365b  prefixlen 64  scopeid 0x20<link>
        ether bc:f6:85:03:36:5b  txqueuelen 1000  (Ethernet)
        RX packets 1171  bytes 148200 (144.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 981  bytes 302750 (295.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

I'm expecting you to solve my problem，thank you very much

mh4x0f, 2018-03-22:
1.  when you trying open the websites, which response the browser return for you ?.
2. did you try use the another DNS server (change this option in File->Settings->General?

MrBeanc, 2018-03-22:
i try the another DNS but it still don't work.The browser has been showing the TLS handshake.

MrBeanc, 2018-03-22:
that is my iptables
[image]

MrBeanc, 2018-03-22:
i think it maybe dns problem,because the Chrome show The IP address of the server can not be found

mh4x0f, 2018-03-26:
3 -  try change the dhcp for isc-dhcp-server in settings, the dhcpd have a proprietary dns server if this not work, sorry but this is not a problem the tool.

closed 2018-04-29 by mh4x0f