P0cL4bs/WiFi-Pumpkin-deprecated #355: New Logo Design Offer as Open Source Contribution

anharismail, 2018-04-14:
Hello Sir. I'm a UI/UX and Graphics Designer. I want propose a new logo for you. Would you mind if I propose new logo for your application as my Open Source Contribution?  Thanks before.

mh4x0f, 2018-04-14:
hi, would be nice, great.

anharismail, 2018-04-15:
@mh4x0f are you the repositories owner? 

I'll post my logo design proposal here as soon as possible

mh4x0f, 2018-04-15:
yes, I'm main coder. wait...

anharismail, 2018-04-19:
Hello Sir, this are my logo design proposals for your app. I'll provide all editable files and all sizes you needed.

Thanks before and hope you like it.

Version 1.
[image]

Version 2.
[image]

mh4x0f, 2018-04-29:
I don't like it, I don't know if i have change the logo again. I'll leave the current logo. thanks

closed 2018-04-29 by mh4x0f