P0cL4bs/WiFi-Pumpkin-deprecated #357: TCP Proxy error

vanika, 2018-04-19:
Hello, cannot resolve this issue, can you tell me wha's wrong with tcp proxy?

OS: Ubuntu 17
Python: 2.7, 3*

```
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/tcp/intercept.py", line 43, in run
    self.main()
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/tcp/intercept.py", line 86, in main
    q = Queue.Queue()
AttributeError: class Queue has no attribute 'Queue'
```

rdbox, 2018-04-29:
I also have a problem kali 4.14.0-kali3-amd64

rdbox, 2018-04-29:
error after the update overloaded the system

**root@kali:~# wifi-pumpkin**
```
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 37, in <module>
    from core.utility.application import ApplicationLoop
  File "/usr/share/WiFi-Pumpkin/core/utility/application.py", line 3, in <module>
    from core.main import version
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 24, in <module>
    from core.utils import (
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 9, in <module>
    from scapy.all import *
  File "/usr/local/lib/python2.7/dist-packages/scapy/all.py", line 11, in <module>
    from scapy.config import *
  File "/usr/local/lib/python2.7/dist-packages/scapy/config.py", line 344, in <module>
    class Conf(ConfClass):
  File "/usr/local/lib/python2.7/dist-packages/scapy/config.py", line 450, in Conf
    crypto_valid = isCryptographyValid()
  File "/usr/local/lib/python2.7/dist-packages/scapy/config.py", line 300, in isCryptographyValid
    return LooseVersion(cryptography.__version__) >= LooseVersion("1.7")
AttributeError: 'module' object has no attribute '__version__'```

closed 2018-04-29 by mh4x0f

Theroxat, 2018-04-30:
Hello,

Kali Rolling
Raspberry pi 3b
Python 2.7.15rc1

Patch applied : https://github.com/P0cL4bs/WiFi-Pumpkin/commit/f34bf650fa5c6b73af3a20b814328e3956aece7e

But same error for me :
```
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/tcp/intercept.py", line 43, in run
    self.main()

  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/tcp/intercept.py", line 86, in main
    q = Queue.Queue()

AttributeError: class Queue has no attribute 'Queue'
```

mh4x0f, 2018-04-30:
@Theroxat this is a problem of version python 2.7, my python 2.7.14 work fine the module Queue has been removed in python 2.7.15. **I'll fix this**. it very strange because in doc the multiprocessing  module the Queue is callable, on python 2.7.14 is not.

vanika, 2018-04-30:
**mh4x0f** My python version is 2.7.14 but still got the same error that **Theroxat** has

mh4x0f, 2018-04-30:
I'll do somes tests.... wait

mh4x0f, 2018-04-30:
check if work fine now and send me feedback.

vanika, 2018-04-30:
``[*] TCPProxy running on port 80/8080:

TCPProxy::ftp               status:On
TCPProxy::hexdump           status:On
TCPProxy::kerberos          status:On
TCPProxy::httpCap           status:On
TCPProxy::imageCap          status:On
TCPProxy::summary           status:On
TCPProxy::emails            status:On

Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/tcp/intercept.py", line 43, in run
    self.main()
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/tcp/intercept.py", line 97, in main
    except Queue.Empty:
AttributeError: class Queue has no attribute 'Empty'

vanika, 2018-04-30:
`[*]` TCPProxy running on port 80/8080:

TCPProxy::ftp               status:On
TCPProxy::hexdump           status:On
TCPProxy::kerberos          status:On
TCPProxy::httpCap           status:On
TCPProxy::imageCap          status:On
TCPProxy::summary           status:On
TCPProxy::emails            status:On

Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/tcp/intercept.py", line 43, in run
    self.main()
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/tcp/intercept.py", line 97, in main
    except Queue.Empty:
AttributeError: class Queue has no attribute 'Empty'
-------------------------------
AP::[FreeZone] Running...
AP::BSSID::[bc:f6:85:03:36:5b] CH `11`

vanika, 2018-04-30:
Got this message: 

`Traceback (most recent call last):
  File "wifi-pumpkin.py", line 37, in <module>
    from core.utility.application import ApplicationLoop
  File "/usr/share/WiFi-Pumpkin/core/utility/application.py", line 3, in <module>
    from core.main import version
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 56, in <module>
    from core.servers.proxy.tcp.intercept import ThreadSniffingPackets
ImportError: No module named tcp.intercept
`

mh4x0f, 2018-04-30:
@vanika lol lol, redownload the tool. I don't know anything about this last traceback

Theroxat, 2018-04-30:
I reinstall the tool from Scratch, now it's work for me without error message.

Many Thank's

vanika, 2018-04-30:
**mh4x0f** yah it was my fault )) now it works normal but sometime shows this error:

`Unhandled Error
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/twisted/python/log.py", line 103, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/usr/local/lib/python2.7/dist-packages/twisted/python/log.py", line 86, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/usr/local/lib/python2.7/dist-packages/twisted/python/context.py", line 122, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/local/lib/python2.7/dist-packages/twisted/python/context.py", line 85, in callWithContext
    return func(*args,**kw)
--- <exception caught here> ---
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 614, in _doReadOrWrite
    why = selectable.doRead()
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 205, in doRead
    return self._dataReceived(data)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 211, in _dataReceived
    rval = self.protocol.dataReceived(data)
  File "/usr/local/lib/python2.7/dist-packages/twisted/protocols/basic.py", line 578, in dataReceived
    why = self.rawDataReceived(data)
  File "/usr/local/lib/python2.7/dist-packages/twisted/web/http.py", line 633, in rawDataReceived
    self.handleResponseEnd()
  File "/usr/share/WiFi-Pumpkin/plugins/external/sslstrip/ServerConnection.py", line 133, in handleResponseEnd
    HTTPClient.handleResponseEnd(self)
  File "/usr/local/lib/python2.7/dist-packages/twisted/web/http.py", line 596, in handleResponseEnd
    self.handleResponse(b)
  File "/usr/share/WiFi-Pumpkin/plugins/external/sslstrip/ServerConnection.py", line 166, in handleResponse
    self.shutdown()
  File "/usr/share/WiFi-Pumpkin/plugins/external/sslstrip/ServerConnection.py", line 203, in shutdown
    self.client.finish()
  File "/usr/local/lib/python2.7/dist-packages/twisted/web/http.py", line 992, in finish
    "Request.finish called on a request after its connection was lost; "
exceptions.RuntimeError: Request.finish called on a request after its connection was lost; use Request.notifyFinish to keep track of this.
`
thanks!!