P0cL4bs/WiFi-Pumpkin-deprecated #360: "wifi-pumpkin.py", line 37, in <module>

rafaleon69, 2018-04-25:
When i run on Kali linux 2018 :wifi-pumpkin
I get 
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 37, in <module>
    from core.utility.application import ApplicationLoop
  File "/usr/share/WiFi-Pumpkin/core/utility/application.py", line 3, in <module>
    from core.main import version
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 24, in <module>
    from core.utils import (
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 9, in <module>
    from scapy.all import *
  File "/usr/local/lib/python2.7/dist-packages/scapy/all.py", line 10, in <module>
    from .config import *
  File "/usr/local/lib/python2.7/dist-packages/scapy/config.py", line 11, in <module>
    from .data import *
  File "/usr/local/lib/python2.7/dist-packages/scapy/data.py", line 184, in <module>
    TCP_SERVICES,UDP_SERVICES=load_services("/etc/services")
  File "/usr/local/lib/python2.7/dist-packages/scapy/data.py", line 109, in load_services
    f=open(filename, errors='ignore')
TypeError: 'errors' is an invalid keyword argument for this function

I tried both python and deb installation, can somebody shed some light on this one ?

All the best.

evaexe, 2018-04-25:
try pip install --upgrade scapy

closed 2018-04-29 by mh4x0f

Tulsipada, 2018-09-02:
Same problems please help