P0cL4bs/WiFi-Pumpkin-deprecated #363: Import Error?

planetxort, 2018-05-05:
I've downloaded wifi-pumpkin and I satisfy all installation requirements listed in the .txt, but for whatever reason i'm getting this strange error message and after three days of trying to troubleshoot, i'm finally turning here for some help..

"You do not have a working installation of the service_identity module: 'cannot import name opentype'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 37, in <module>
    from core.utility.application import ApplicationLoop
  File "/usr/share/WiFi-Pumpkin/core/utility/application.py", line 3, in <module>
    from core.main import version
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 55, in <module>
    from netfilterqueue import NetfilterQueue
ImportError: /usr/local/lib/libnetfilter_queue.so.1: undefined symbol: mnl_attr_parse"

Does anyone know how I can fix this? I'm struggling.

BassHunter200, 2018-05-06:
Same problem here , I believe it is because it isn't configured to run python3 which means Wifi-Pumkin needs an update or something

planetxort, 2018-05-06:
Damn...

closed 2018-05-14 by mh4x0f