P0cL4bs/WiFi-Pumpkin-deprecated #389: can not create fake AP

rdbox, 2018-07-20:
## What's the problem (or question)?

the problem is I'm running WiFi-Pumpkin I'm pushing start a new AP is being created but I can not connect to it.

I can not connect if I put a password or without a password anyway the connection on the device resets the error of the proxy with authentication.

I tried to connect via wifi laptop is not working
I tried to connect through the phone androyd does not work
Error with identical authorization problems

I also tried changing the DHCP settings to specify different ip subnets.
This also did not help
 
#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: 
* Virtual Machine (yes or no and which): 
* Operating System and version: 

1. **I'm working in Virtualbox guest system**
2. **v 0.8.5** 
3. **wifi** 

```phy0	wlan0		rt2800usb	D-Link Corp. DWA-125 Wireless N 150 Adapter(rev.A3) [Ralink RT5370]```

[image]

rdbox, 2018-07-20:
Are you sure exactly? I will add notes from myself my wifi adapter is connected via usb and then I transferred it to the guest VirtualBox

maybe there is a solution? other than how to use vmware it requires a lot of resources is bad

mh4x0f, 2018-07-21:
#43 look this issue, a lot people have problem only bridge mode in virtualbox.

closed 2018-07-21 by mh4x0f