P0cL4bs/WiFi-Pumpkin-deprecated #394: tcpdump: wlan1: That device is not up

nohupt, 2018-07-27:
## What's the problem (or question)?

When starting the AP, I get an error telling me that my interface is not up even though it is. Here is the output: 

```
Loading GUI...
WiFi-Pumpkin Running!

[*] Loading debugging mode
[*] Current Session::ID [NzI1MjM=]
[*] Configuring hostapd...
[*] Enable forwarding in iptables...
[*] Configuring dhcpd...
[*] Sharing Internet Connections with NAT...
[New Thread 8993 (hostapd)]
[New Thread 8994 (DNSServer)]
[*] Pumpkin-Proxy running on port:8080 

[*] TCPProxy running on port 80/8080:

TCPProxy::ftp               status:On
TCPProxy::hexdump           status:On
TCPProxy::kerberos          status:On
TCPProxy::httpCap           status:On
TCPProxy::imageCap          status:On
TCPProxy::summary           status:On
TCPProxy::emails            status:On

-------------------------------
AP::[blabbity] Running...
tcpdump: wlan1: That device is not up
AP::BSSID::[ae:2a:da:94:f1:47] CH 6
-------------------------------
Thread::[hostapd] successfully stopped.
Thread::[DNSServer] successfully stopped.
tcpdump: wlan1: That device is not up
tcpdump: wlan1: That device is not up
tcpdump: wlan1: That device is not up
tcpdump: wlan1: That device is not up
tcpdump: wlan1: That device is not up
tcpdump: wlan1: That device is not up
Thread::[Pumpkin-Proxy] successfully stopped.
Thread::[TCPProxy] successfully stopped.
tcpdump: wlan1: That device is not up
```
And this error shows on the Wifi Pumpkin screen:
[image]

wlan1 is up and runs fine to connect to other APs, and is fine when switching to monitor mode and using airbase. 

I also tested tcpdump and hostapd outside Pumpkin and they run fine, no errors. 

What I don't understand is that I ran Pumpkin a few days ago with the exact same set up and had no problems. Could this be a network configuration that somehow got changed due to a forgotten OS update/sleepiness/drunkeness?

This makes no sense! Help!

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): 
     **ALFA AWUS036NEH -- yes, supports AP**
* Version used tool: 
     **0.8.5**
* Virtual Machine (yes or no and which):  
     **NO**
* Operating System and version: 
     **Kali Linux Rolling**

closed 2018-07-29 by mh4x0f

nohupt, 2018-07-30:
You could at least link to the duplicate

mh4x0f, 2018-07-30:
> What I don't understand is that I ran Pumpkin a few days ago with the exact same set up and had no problems. Could this be a network configuration that somehow got changed due to a forgotten OS update/sleepiness/drunkeness?

yes, probably the update changed the setting network configuration. the error come from hostapd that can't mount the AP. #184