P0cL4bs/WiFi-Pumpkin-deprecated #403: AttributeError: class Queue has no attribute 'Queue'

ntnychss, 2018-08-29:
Before opening a new issue, please read this [wiki](https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Bug-reporting)

## What's the problem (or question)?

I've seen that the problem as been submitted yet but it still persists (at least with my version of python)

> ➜  ~ sudo wifi-pumpkin
Loading GUI...
WiFi-Pumpkin Running!
Starting Thread:Thread Scanner AP::scapy
2.7.12 (default, Dec  4 2017, 14:50:18)  <--- I modified your code to print what version was used
[GCC 5.4.0 20160609]
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/packets/wireless.py", line 38, in run
    self.LoopScanmer()
  File "/usr/share/WiFi-Pumpkin/core/packets/wireless.py", line 51, in LoopScanmer
    q = Queue.Queue()
AttributeError: class Queue has no attribute 'Queue'

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): Built in and also with my Alpha
* Version used tool: Last one (just cloned the project and manually installed it)
* Virtual Machine (yes or no and which):  None
* Operating System and version: Ubuntu 16.04.5

Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.5 LTS
Release:	16.04
Codename:	xenial

I have tried everything: unistalling it - reinstalling it, unistall and reinstall python; tried the .deb
An interesting fact is that if I run the same code from terminal it works.

> ➜  ~ python   
Python 2.7.12 (default, Dec  4 2017, 14:50:18) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
\>\>\> import queue as Queue
\>\>\> q = Queue.Queue()
\>\>\> import sys
\>\>\> sys.version
'2.7.12 (default, Dec  4 2017, 14:50:18) \n[GCC 5.4.0 20160609]'

Thanks in advance!

mh4x0f, 2018-09-13:
Nice report, brown I'll check this coming soon, but the deauth module need to update.

mh4x0f, 2018-09-15:
@ntnychss did you try update  python version  to 2.7.14 ?

ntnychss, 2018-09-17:
@mh4x0f I remember that I tried to install a different version of python < than my current one, but unfortunately I don't remember which one. You can use a  Ubuntu 16.04 VM and try to install and use it there. In my distro I only installed required updates so it should be enough to replicate the bug environment.

mh4x0f, 2018-09-17:
@ntnychss ok nice, I'll check this. there are a option for use airodump-ng on settings if you want to use the feature.

mh4x0f, 2018-10-08:
@ntnychss try to update the python to 2.7.14, this is the solution for now. thank's

yckai, 2018-11-03:
Hi, i have a bug with Wifi Pumpkin, when i start arp poisoner i have this error : 

`Loading GUI...
WiFi-Pumpkin Running!
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Starting Thread:Packets SpoofStarting Thread:Arp Poison:: [gateway]
 
Starting Thread:Arp::Poison => [target]
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/packets/network.py", line 134, in run
    self.sniff()
  File "/usr/share/WiFi-Pumpkin/core/packets/network.py", line 158, in sniff
    q = Queue.Queue()
AttributeError: class Queue has no attribute 'Queue'
`
**Environment**
- Python 2.7.15+
- wifi-pumpkin is up to date
- Kali linux  last version
- Vmware workstation
- Network adaptateur bridge 

Thanks in advance!