P0cL4bs/WiFi-Pumpkin-deprecated #421: Wireless security is OFF but asking for password

GGGunrunner, 2018-10-30:
Raspberry Pi 3
Kali 2018
TP-Link 722N
Latest Pumpkin git

I've got it installed, no errors. Ran for the first time, all default settings.
AP Name: PumpAP
Security Password: OFF

But whenever I try to connect with my iphone, ipad, or android, it asks for the password as if there were WPA PSK set.... Strange. Anyone else got this bug? Thanks!

mh4x0f, 2018-10-31:
send me the output the command bellow:
``` sh 
cat /usr/share/WiFi-Pumpkin/core/config/hostapd/hostapd.conf
```

GGGunrunner, 2018-11-01:
That bug has fixed itself - I didnt pay attention that it was wlan0 by default (the onboard chip of Raspberry Pi3 which has no monitor mode).

 After selecting wlan1 (external TPLink), it tries to lauch the AP, writes :

[*] Loading debugging mode
[*] Current Session::ID [MjYyNzI=]
[*] Configuring hostapd...

And throws an error "Errno 2: No such file or directory" ontop the Pumpkin window.
The full install went fine so I really dont know what else is missing.
If there any chance to debug which file it tries to open?
Via lsof or strace? In Windows, its simple (Sys Internals tools), but does anyone know how to monitor file I/O in linux? Thanks all!

closed 2018-11-02 by mh4x0f