P0cL4bs/WiFi-Pumpkin-deprecated #423: website design example

mh4x0f, 2018-10-31:
hi all dev front-end user, I want to create an website for WiFi-Pumpkin doc. please send me suggestions. 
colors: based on WiFi-Pumpkin 
style: bootstrap

minhducsun2002, 2019-01-10:
1. Isn't GitHub Wiki enough?
2. 
```
index.html
|-docs
  |-Getting started
  |-Installation
  |-PumpkinProxy
  |-

minhducsun2002, 2019-01-10:
1. Isn't GitHub Wiki enough?
2. 
```
index.html
|-docs
  |-Getting started
  |-Installation
  |-PumpkinProxy
  |-TCPProxy
|-FAQs
```
3. About docs, it should be possible that the underlying logic be explained? Like, the reason feature A doesn't work when feature B is active, for example.

minhducsun2002, 2019-01-11:
> I do not understand, tell me an example.

I mean, something like "If `NetworkManager` shows that Wi-Fi isn't managed even when WiFi-Pumpkin is not active, try `systemctl restart NetworkManager` and see if it works for you. That is because behind the scenes WiFi-Pumpkin uses `hostapd` which may conflict with `NetworkManager`.". Like tips and FAQs, but where necessary.

mh4x0f, 2019-01-11:
ok, cool

m1sf1t42, 2019-01-24:
Hey there @mh4x0f , 
Do you already got any suggestions?
If not, I would send you one the next days, got nothing to do and I'd like to support the project.

mh4x0f, 2019-01-27:
> Do you already got any suggestions?

No, not yet

jeretc, 2019-03-19:
> hi all dev front-end user, I want to create an website for WiFi-Pumpkin doc. please send me suggestions.
> colors: based on WiFi-Pumpkin
> style: bootstrap
> build: github pages

hi i think for documentation, you could use this https://readthedocs.org/
check out some of the sample projects.

closed 2019-04-09 by mh4x0f