P0cL4bs/WiFi-Pumpkin-deprecated #424: AttributeError: class Queue has no attribute 'Queue'

yckai, 2018-11-03:
Before opening a new issue, please read this [wiki](https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Bug-reporting)

## What's the problem (or question)?

*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: 
* Virtual Machine (yes or no and which): 
* Operating System and version:

yckai, 2018-11-03:
Hi, i have a bug with Wifi Pumpkin, when i start arp poisoner i have this error :

`Loading GUI...
WiFi-Pumpkin Running!
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Starting Thread:Packets SpoofStarting Thread:Arp Poison:: [gateway]

Starting Thread:Arp::Poison => [target]
Traceback (most recent call last):
File "/usr/share/WiFi-Pumpkin/core/packets/network.py", line 134, in run
self.sniff()
File "/usr/share/WiFi-Pumpkin/core/packets/network.py", line 158, in sniff
q = Queue.Queue()
AttributeError: class Queue has no attribute 'Queue'
`
Environment

Python 2.7.15+
wifi-pumpkin is up to date
Kali linux last version
Vmware workstation
Network adaptateur bridge
Thanks in advance!

howto5166, 2018-11-03:
When i try to run wifi-pumpkin. I get the following error...
user@lubuntu:~$ sudo wifi-pumpkin 
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 35, in <module>
    from core.loaders.checker.depedences import check_dep_pumpkin
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/depedences.py", line 2, in <module>
    from core.utility.collection import SettingsINI
  File "/usr/share/WiFi-Pumpkin/core/utility/collection.py", line 2, in <module>
    from PyQt4.QtCore import QSettings
ImportError: No module named PyQt4.QtCore
I'm using lubuntu latest version with fully upgraded source code..

closed 2018-11-04 by mh4x0f