P0cL4bs/WiFi-Pumpkin-deprecated #428: pumpkin-proxy greyed out

hdawg12, 2018-11-28:
Before opening a new issue, please read this [wiki](https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Bug-reporting)

## What's the problem (or question)?
Pumpkin_proxy is not working for me.  It's greyed out.  I believe I updated  mitmproxy 0.18.2. I also tried to reinistall wifi-pumpkin, but it still has the same issue.

*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): Atheros, ath9k , yest it supports AP mode
* Version used tool: 
* Virtual Machine (yes or no and which): Yes, vmware workstation
* Operating System and version:  Kali linux 2018

JunyuT, 2018-11-29:
Same thing here. I've looked through all the previous issues, installed and reinstalled a bunch of packages, messed around with different distros to get mitmproxy 0.18.2 to work, but it never shows in wifi-pumpkin.

hdawg12, 2018-11-29:
which packages did you reinstall?

hdawg12, 2018-11-29:
I think I was able to install mitmproxy 0.18.2, but I don't know if I have to activate it before I activate wifi-pumpkin or if it detects the software by itself.  I'm not sure how to use mitmproxy yet.

JunyuT, 2018-11-29:
run `mitmproxy` to see if it works, and `mitmdump --version` to verify the correct version is selected. 

I had to reinstall mitmproxy, libssl and a bunch of others. If you look at the closed issues, some of them will point you to what you need to look at.

hdawg12, 2018-11-30:
My mitmproxy works and my mitmdump says 0.18.2.  But I still have the part greyed out.  Do you think I need to reinstall wifi-pumpkin?

JunyuT, 2018-11-30:
No idea. I'm running into the same problem, as I said.

hdawg12, 2018-11-30:
im reading through some of the closed issues hoping it would provide me with an answer. If you find it, let me know. I"ll do the same.

mh4x0f, 2018-11-30:
the run this command to check if the mitmproxy correct version and send me the output.
```sh
python --version && pip freeze | grep "mitmproxy"
```

hdawg12, 2018-11-30:
@mh4x0f when I run that command, I get this output:

Python 2.7.15+
mitmproxy==0.18.2

mh4x0f, 2018-12-04:
@hdawg12 send me a screenshot, I'm don't understand because not working for you. the WP check the version is working maybe the mitmproxy is not installed correctly. execute this command bellow and send  me the output again.
``` sh 
python -c "from mitmproxy import proxy, flow, options "
python -c "from mitmproxy.proxy.server import ProxyServer"
```
obs: if not get error in terminal after executed the command above, the module is working fine.

hdawg12, 2018-12-06:
There was an issue with booting up my kali linux so I had to reinstall everything again.  When I downloaded wifi-pumpkin, It successfully installed pumpkin-proxy.  Not sure what the issue was on my previous version of wifi-pumpkin.

closed 2018-12-06 by mh4x0f