P0cL4bs/WiFi-Pumpkin-deprecated #429: NameError: name 'dir_path' is not defined

makejohn, 2018-11-30:
running problem
OS: kali linux install on Macbook pro

**problem:**

Traceback (most recent call last):
  File "wifi-pumpkin.py", line 29, in <module>
    import core.utility.constants as C
  File "/usr/share/WiFi-Pumpkin/core/utility/constants.py", line 72, in <module>
    TEMP_CUSTOM = dir_path+'/templates/phishing/custom'
NameError: name 'dir_path' is not defined

**I try to define dir_path  = '/usr/share/WiFi-Pumpkin'
but then, new error message shows:**

Traceback (most recent call last):
  File "wifi-pumpkin.py", line 36, in <module>
    from core.loaders.checker.networkmanager import CLI_NetworkManager,UI_NetworkManager
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 3, in <module>
    from core.main import Initialize,QtGui
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 17, in <module>
    from core.utility.threads import ThreadReactor,ThreadPopen
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 20, in <module>
    from core.servers.proxy.http.controller.handler import MasterHandler
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/http/controller/handler.py", line 1, in <module>
    from plugins.extension import *
  File "/usr/share/WiFi-Pumpkin/plugins/extension/stickycookie.py", line 1, in <module>
    from mitmproxy.models import decoded
  File "/usr/local/lib/python2.7/dist-packages/mitmproxy/models/__init__.py", line 4, in <module>
    from .connections import ClientConnection, ServerConnection
  File "/usr/local/lib/python2.7/dist-packages/mitmproxy/models/connections.py", line 11, in <module>
    from netlib import tcp
  File "/usr/local/lib/python2.7/dist-packages/netlib/tcp.py", line 18, in <module>
    from backports import ssl_match_hostname
ImportError: cannot import name ssl_match_hostname

mh4x0f, 2018-12-04:
remove the mitmproxy plugin command bellow:
``` sh 
pip uninstall mitmproxy==0.18.2
```

makejohn, 2018-12-06:
Thanks, problem solved!

closed 2018-12-06 by mh4x0f