P0cL4bs/WiFi-Pumpkin-deprecated #435: Old AP name stays the same when discovered by PC

hdawg12, 2018-12-17:
Before opening a new issue, please read this [wiki](https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Bug-reporting)

## What's the problem (or question)?

When I rename my AP and restart wifi-pumpkin, I login to my other pc and I see the old AP name. For example, if I old name is "free wifi" and I go into Wifi-pumpkin and change name to "Don Wifi", when I go into my other laptop, I see "free wifi" as one of the AP's, not Don Wifi. 
Is there a way to fix this?

*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):Panda Wireless
* Version used tool: Wifi-Pumpkin 0.8.7
* Virtual Machine (yes or no and which): nope
* Operating System and version: Kali linux 2018 Rolling

mh4x0f, 2018-12-21:
nice report, I'll be check this coming soon.

hdawg12, 2018-12-22:
@mh4x0f , great.  Let me know how it goes.  Thanks

hdawg12, 2018-12-27:
When I startup wifi-pumpkin, I get this new error:
root@Kali2018:~# wifi-pumpkin 
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 36, in <module>
    from core.loaders.checker.networkmanager import CLI_NetworkManager,UI_NetworkManager
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 3, in <module>
    from core.main import Initialize,QtGui
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 17, in <module>
    from core.utility.threads import ThreadReactor,ThreadPopen
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 20, in <module>
    from core.servers.proxy.http.controller.handler import MasterHandler
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/http/controller/handler.py", line 1, in <module>
    from plugins.extension import *
  File "/usr/share/WiFi-Pumpkin/plugins/extension/stickycookie.py", line 1, in <module>
    from mitmproxy.models import decoded
  File "/usr/local/lib/python2.7/dist-packages/mitmproxy/models/__init__.py", line 4, in <module>
    from .connections import ClientConnection, ServerConnection
  File "/usr/local/lib/python2.7/dist-packages/mitmproxy/models/connections.py", line 11, in <module>
    from netlib import tcp
  File "/usr/local/lib/python2.7/dist-packages/netlib/tcp.py", line 18, in <module>
    from backports import ssl_match_hostname
ImportError: cannot import name ssl_match_hostname

Is it because I need to update it since you fixed the issue it was having with the AP?

closed 2018-12-27 by mh4x0f