P0cL4bs/WiFi-Pumpkin-deprecated #436: Error when runing

mother2110, 2018-12-17:
Before opening a new issue, please read this [wiki](https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Bug-reporting)

## What's the problem (or question)?

[root:~/pentesting/wifi/WiFi-Pumpkin]# sudo wifi-pumpkin                                                                   (master✱) 
You do not have a working installation of the service_identity module: 'cannot import name opentype'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 36, in <module>
    from core.loaders.checker.networkmanager import CLI_NetworkManager,UI_NetworkManager
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 3, in <module>
    from core.main import Initialize,QtGui
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 17, in <module>
    from core.utility.threads import ThreadReactor,ThreadPopen
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 20, in <module>
    from core.servers.proxy.http.controller.handler import MasterHandler
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/http/controller/handler.py", line 1, in <module>
    from plugins.extension import *
  File "/usr/share/WiFi-Pumpkin/plugins/extension/shakepage.py", line 1, in <module>
    from mitmproxy.models import decoded
  File "/usr/local/lib/python2.7/dist-packages/mitmproxy/models/__init__.py", line 4, in <module>
    from .connections import ClientConnection, ServerConnection
  File "/usr/local/lib/python2.7/dist-packages/mitmproxy/models/connections.py", line 11, in <module>
    from netlib import tcp
  File "/usr/local/lib/python2.7/dist-packages/netlib/tcp.py", line 18, in <module>
    from backports import ssl_match_hostname
ImportError: cannot import name ssl_match_hostname

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: .8.7
* Virtual Machine (yes or no and which): No
* Operating System and version:  Kali 2018.4

mother2110, 2018-12-17:
I do have service_identity installed per the other responses.

hdawg12, 2018-12-17:
pip install --upgrade google-auth-oauthlib  worked for me

closed 2018-12-18 by mh4x0f

mother2110, 2018-12-18:
Why close it when it has not been resolved? I have tried everything I have read here.

afkalways, 2018-12-20:
> Why close it when it has not been resolved? I have tried everything I have read here.

same situation

mother2110, 2018-12-20:
Ya, I gave up for now on this. Been trying all year. There are other tools out there.

snorx82, 2019-01-11:
same problem ....

"Traceback (most recent call last):
  File "wifi-pumpkin.py", line 36, in <module>
    from core.loaders.checker.networkmanager import CLI_NetworkManager,UI_NetworkManager
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 3, in <module>
    from core.main import Initialize,QtGui
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 33, in <module>
    from core.controllers.proxycontroller import *
  File "/usr/share/WiFi-Pumpkin/core/controllers/proxycontroller.py", line 4, in <module>
    from core.servers.proxy.package import *
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/package/SSLStripProxy.py", line 15, in <module>
    from core.widgets.docks.dockmonitor import (
  File "/usr/share/WiFi-Pumpkin/core/widgets/docks/dockmonitor.py", line 1, in <module>
    from plugins.extension import *
  File "/usr/share/WiFi-Pumpkin/plugins/extension/shakepage.py", line 1, in <module>
    from mitmproxy.models import decoded
ImportError: No module named mitmproxy.models"