P0cL4bs/WiFi-Pumpkin-deprecated #443: ImportError: cannot import name ssl_match_hostname

vishwanathkavya, 2018-12-29:
You do not have a working installation of the service_identity module: 'cannot import name opentype'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 36, in <module>
    from core.loaders.checker.networkmanager import CLI_NetworkManager,UI_NetworkManager
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 3, in <module>
    from core.main import Initialize,QtGui
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 17, in <module>
    from core.utility.threads import ThreadReactor,ThreadPopen
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 20, in <module>
    from core.servers.proxy.http.controller.handler import MasterHandler
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/http/controller/handler.py", line 1, in <module>
    from plugins.extension import *
  File "/usr/share/WiFi-Pumpkin/plugins/extension/downloadspoof.py", line 2, in <module>
    from mitmproxy.models import decoded
  File "/usr/local/lib/python2.7/dist-packages/mitmproxy/models/__init__.py", line 4, in <module>
    from .connections import ClientConnection, ServerConnection
  File "/usr/local/lib/python2.7/dist-packages/mitmproxy/models/connections.py", line 11, in <module>
    from netlib import tcp
  File "/usr/local/lib/python2.7/dist-packages/netlib/tcp.py", line 18, in <module>
    from backports import ssl_match_hostname
ImportError: cannot import name ssl_match_hostname

help me out !! I'm looking for answers !! 

I've previously installed and worked on it. It did got executed without any errors.

closed 2018-12-30 by mh4x0f

tyokinuhata, 2018-12-30:
@vishwanathkavya 
try this command
```
cp -r /usr/local/lib/python2.7/dist-packages/backports/ssl_match_hostname/ /usr/lib/python2.7/dist-packages/backports
```

vishwanathkavya, 2018-12-30:
thanks @tyokinuhata it worked !! GUI opened up but unable to see devices on home and unable to connect to fakeAP. which is no internet. I'm using Alfa awus036neh on a lenovo g510 with kali 2019.1 live.

hdawg12, 2019-01-07:
I don't have backports or ssl_match_hostname

vishwanathkavya, 2019-01-07:
@hdawg12 
Try this !! 
pip install backports.ssl_match_hostname