P0cL4bs/WiFi-Pumpkin-deprecated #448: Urgent ** Help with Pumpkin Failure

Pen-Ace, 2019-01-03:
## What's the problem (or question)?
After running sudo wifi-pumpkin the app dont work.

#### Please tell us details about your environment.
Parrot OS 4.4 
VM 

#### Scenario 1 

When I run: pip install -r requirements.txt all looks to be satisfied:
Requirement already satisfied: Twisted in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 1)) (18.7.0)
Requirement already satisfied: scapy in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 2)) (2.4.0)
Requirement already satisfied: bs4 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 3)) (0.0.1)
Requirement already satisfied: netaddr in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 4)) (0.7.19)
Requirement already satisfied: config in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 5)) (0.4.0)
Requirement already satisfied: dnspython in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 6)) (1.15.0)
Requirement already satisfied: isc_dhcp_leases in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 7)) (0.9.1)
Requirement already satisfied: netifaces in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 8)) (0.10.9)
Requirement already satisfied: pcapy in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 9)) (0.10.8)
Requirement already satisfied: configparser==3.3.0r1 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 10)) (3.3.0.post1)
Requirement already satisfied: NetfilterQueue in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 11)) (0.8.1)
Requirement already satisfied: configobj in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 12)) (5.0.6)
Requirement already satisfied: libarchive-c==2.1 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 13)) (2.1)
Requirement already satisfied: python-magic==0.4.6 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 14)) (0.4.6)
Requirement already satisfied: pefile in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 15)) (2018.8.8)
Requirement already satisfied: capstone in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 16)) (3.0.5)
Requirement already satisfied: hyperframe in /home/odin/.local/lib/python2.7/site-packages (from -r requirements.txt (line 17)) (4.0.2)
Requirement already satisfied: h2 in /home/odin/.local/lib/python2.7/site-packages (from -r requirements.txt (line 18)) (2.4.4)
Requirement already satisfied: scapy_http in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 19)) (1.8.2)
Requirement already satisfied: service_identity in /home/odin/.local/lib/python2.7/site-packages (from -r requirements.txt (line 20)) (18.1.0)
Requirement already satisfied: beautifulsoup4 in /usr/lib/python2.7/dist-packages (from bs4->-r requirements.txt (line 3)) (4.6.3)
Requirement already satisfied: six in /home/odin/.local/lib/python2.7/site-packages (from isc_dhcp_leases->-r requirements.txt (line 7)) (1.10.0)
Requirement already satisfied: enum34<2,>=1.0.4; python_version == "2.7" or python_version == "3.3" in /home/odin/.local/lib/python2.7/site-packages (from h2->-r requirements.txt (line 18)) (1.1.6)
Requirement already satisfied: hpack<3,>=2.2 in /usr/local/lib/python2.7/dist-packages (from h2->-r requirements.txt (line 18)) (2.3.0)
Requirement already satisfied: pyasn1 in /home/odin/.local/lib/python2.7/site-packages (from service_identity->-r requirements.txt (line 20)) (0.4.5)
Requirement already satisfied: pyasn1-modules in /home/odin/.local/lib/python2.7/site-packages (from service_identity->-r requirements.txt (line 20)) (0.2.3)
Requirement already satisfied: ipaddress; python_version < "3.3" in /home/odin/.local/lib/python2.7/site-packages (from service_identity->-r requirements.txt (line 20)) (1.0.22)
Requirement already satisfied: attrs>=16.0.0 in /home/odin/.local/lib/python2.7/site-packages (from service_identity->-r requirements.txt (line 20)) (18.2.0)
Requirement already satisfied: cryptography in /home/odin/.local/lib/python2.7/site-packages (from service_identity->-r requirements.txt (line 20)) (1.5.3)
Requirement already satisfied: cffi>=1.4.1 in /home/odin/.local/lib/python2.7/site-packages (from cryptography->service_identity->-r requirements.txt (line 20)) (1.11.5)
Requirement already satisfied: setuptools>=11.3 in /usr/lib/python2.7/dist-packages (from cryptography->service_identity->-r requirements.txt (line 20)) (40.6.2)
Requirement already satisfied: idna>=2.0 in /home/odin/.local/lib/python2.7/site-packages (from cryptography->service_identity->-r requirements.txt (line 20)) (2.8)
Requirement already satisfied: pycparser in /home/odin/.local/lib/python2.7/site-packages (from cffi>=1.4.1->cryptography->service_identity->-r requirements.txt (line 20)) (2.19)

#### Scenario 2 
However, $sudo wifi-pumpkin gives me this Error:
You do not have a working installation of the service_identity module: 'cannot import name opentype'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 36, in <module>
    from core.loaders.checker.networkmanager import CLI_NetworkManager,UI_NetworkManager
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 3, in <module>
    from core.main import Initialize,QtGui
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 17, in <module>
    from core.utility.threads import ThreadReactor,ThreadPopen
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 20, in <module>
    from core.servers.proxy.http.controller.handler import MasterHandler
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/http/controller/handler.py", line 1, in <module>
    from plugins.extension import *
  File "/usr/share/WiFi-Pumpkin/plugins/extension/beef.py", line 1, in <module>
    from mitmproxy.models import decoded
ImportError: No module named mitmproxy.models

Please can you help.

The12MT, 2019-01-03:
Same problem. Tried it with parrot (no vm and not live) and kali (vm) both giving this error

hdawg12, 2019-01-07:
Has anyone figured this out?

101zombie101, 2019-01-07:
@hdawg12

Still waiting to hear back.

hdawg12, 2019-01-07:
I tried this:

pip install service_identity
pip install --upgrade pyasn1-modules

which seemed to resolve the issue for me.

101zombie101, 2019-01-07:
I'll give that a try. Thanks. @hdawg12 
I'll let you know if it worked for me.

101zombie101, 2019-01-08:
@hdawg12
Did not work for me.

Pen-Ace, 2019-01-10:
@hdawg12 This did not work for me either. 

pip install service_identity
Requirement already satisfied: service_identity in /home/X/.local/lib/python2.7/site-packages (18.1.0)
Requirement already satisfied: pyasn1-modules in /home/X/.local/lib/python2.7/site-packages (from service_identity) (0.2.3)
Requirement already satisfied: ipaddress; python_version < "3.3" in /home/X/.local/lib/python2.7/site-packages (from service_identity) (1.0.22)
Requirement already satisfied: attrs>=16.0.0 in /home/X/.local/lib/python2.7/site-packages (from service_identity) (18.2.0)
Requirement already satisfied: pyasn1 in /home/X/.local/lib/python2.7/site-packages (from service_identity) (0.4.5)
Requirement already satisfied: cryptography in /home/X/.local/lib/python2.7/site-packages (from service_identity) (1.5.3)
Requirement already satisfied: six>=1.4.1 in /home/X/.local/lib/python2.7/site-packages (from cryptography->service_identity) (1.10.0)
Requirement already satisfied: cffi>=1.4.1 in /home/X/.local/lib/python2.7/site-packages (from cryptography->service_identity) (1.11.5)
Requirement already satisfied: setuptools>=11.3 in /usr/lib/python2.7/dist-packages (from cryptography->service_identity) (40.6.2)
Requirement already satisfied: enum34 in /home/X/.local/lib/python2.7/site-packages (from cryptography->service_identity) (1.1.6)
Requirement already satisfied: idna>=2.0 in /home/X/.local/lib/python2.7/site-packages (from cryptography->service_identity) (2.8)
Requirement already satisfied: pycparser in /home/X/.local/lib/python2.7/site-packages (from cffi>=1.4.1->cryptography->service_identity) (2.19)

pip install --upgrade pyasn1-modules
Requirement already up-to-date: pyasn1-modules in /home/X/.local/lib/python2.7/site-packages (0.2.3)
Requirement already satisfied, skipping upgrade: pyasn1<0.5.0,>=0.4.1 in /home/X/.local/lib/python2.7/site-packages (from pyasn1-modules) (0.4.5)

Any help on getting this working would be appreciated. FYI - i have also tested in a fresh install of WFP on Parrot OS.

minhducsun2002, 2019-01-11:
Sounds like a problem with `mitmproxy`. See my issue, #440.

closed 2019-01-11 by mh4x0f

101zombie101, 2019-01-11:
Just uninstall and reinstall WiFi Pumpkin and it should fix the problem? Correct?

hdawg12, 2019-01-12:
how do u uninstall wifi pumpkin

minhducsun2002, 2019-01-12:
Don't we have `sudo ./uninstall.sh`?

minhducsun2002, 2019-01-12:
> Reinstalled it and still no luck. What am I doing wrong?

Execute `wifi-pumpkin` in your shell with superuser privilege and paste the output here.

Pen-Ace, 2019-01-12:
Update - OS 2:
I have the same problem with a fresh install using Kodachi:

[17:31:46] kodachi@Live-OS:~/Desktop/WiFi-Pumpkin $ sudo wifi-pumpkin
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 36, in <module>
    from core.loaders.checker.networkmanager import CLI_NetworkManager,UI_NetworkManager
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 3, in <module>
    from core.main import Initialize,QtGui
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 33, in <module>
    from core.controllers.proxycontroller import *
  File "/usr/share/WiFi-Pumpkin/core/controllers/proxycontroller.py", line 4, in <module>
    from core.servers.proxy.package import *
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/package/TCP-Proxy.py", line 17, in <module>
    from core.widgets.docks.dockmonitor import (
  File "/usr/share/WiFi-Pumpkin/core/widgets/docks/dockmonitor.py", line 1, in <module>
    from plugins.extension import *
  File "/usr/share/WiFi-Pumpkin/plugins/extension/stickycookie.py", line 1, in <module>
    from mitmproxy.models import decoded
ImportError: No module named mitmproxy.models

What is going on!!! @mh4x0f @101zombie101 @The12MT - going to try Kali now

Pen-Ace, 2019-01-12:
OS 4 - Parrot 4.4

The same process does not work for Parrot im afraid 

sudo wifi-pumpkin
You do not have a working installation of the service_identity module: 'cannot import name opentype'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 36, in <module>
    from core.loaders.checker.networkmanager import CLI_NetworkManager,UI_NetworkManager
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 3, in <module>
    from core.main import Initialize,QtGui
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 33, in <module>
    from core.controllers.proxycontroller import *
  File "/usr/share/WiFi-Pumpkin/core/controllers/proxycontroller.py", line 4, in <module>
    from core.servers.proxy.package import *
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/package/BDFProxy.py", line 16, in <module>
    from core.widgets.docks.dockmonitor import (
  File "/usr/share/WiFi-Pumpkin/core/widgets/docks/dockmonitor.py", line 1, in <module>
    from plugins.extension import *
  File "/usr/share/WiFi-Pumpkin/plugins/extension/beef.py", line 1, in <module>
    from mitmproxy.models import decoded
ImportError: No module named mitmproxy.models

Not sure why.

minhducsun2002, 2019-01-13:
@Pen-Ace Are you sure you installed the Python 2 version of `mitmproxy`? Try `pip2 install mitmprxoxy==0.18.2`.

In case of failure, try the latest installaion shell script. It contains a fix for that.

Pen-Ace, 2019-01-13:
Hi @minhducsun2002,

Ran that in Parrot OS 4.4: pip2 install mitmproxy==0.18.2 and it fails building the wheel for cryptography with this output when done:

Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-3kgoCm/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-Mx0Med/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-install-3kgoCm/cryptography/

Then tried:
sudo wifi-pumpkin
You do not have a working installation of the service_identity module: 'cannot import name opentype'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 36, in <module>
    from core.loaders.checker.networkmanager import CLI_NetworkManager,UI_NetworkManager
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 3, in <module>
    from core.main import Initialize,QtGui
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 33, in <module>
    from core.controllers.proxycontroller import *
  File "/usr/share/WiFi-Pumpkin/core/controllers/proxycontroller.py", line 4, in <module>
    from core.servers.proxy.package import *
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/package/BDFProxy.py", line 16, in <module>
    from core.widgets.docks.dockmonitor import (
  File "/usr/share/WiFi-Pumpkin/core/widgets/docks/dockmonitor.py", line 1, in <module>
    from plugins.extension import *
  File "/usr/share/WiFi-Pumpkin/plugins/extension/beef.py", line 1, in <module>
    from mitmproxy.models import decoded
ImportError: No module named mitmproxy.models

Any ideas?

mh4x0f, 2019-01-13:
try this:
``` sh
sudo apt-get install gdebi
wget http://ftp.de.debian.org/debian/pool/main/m/mitmproxy/mitmproxy_0.18.2-6_all.deb \
		-O /tmp/mitmproxy_0.18.2-6_all.deb
gdebi --non-interactive /tmp/mitmproxy_0.18.2-6_all.deb
```

minhducsun2002, 2019-01-14:
@Pen-Ace, please see the suggestion from the maintainer, @mh4x0f.

101zombie101, 2019-01-14:
@Pen-Ace 

Please let us know if this worked and the steps please. Thank you.

Pen-Ace, 2019-01-15:
Hi @mh4x0f @101zombie101 

i tried your input and this was the error when running WP:

Traceback (most recent call last):
  File "wifi-pumpkin.py", line 36, in <module>
    from core.loaders.checker.networkmanager import CLI_NetworkManager,UI_NetworkManager
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 3, in <module>
    from core.main import Initialize,QtGui
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 33, in <module>
    from core.controllers.proxycontroller import *
  File "/usr/share/WiFi-Pumpkin/core/controllers/proxycontroller.py", line 4, in <module>
    from core.servers.proxy.package import *
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/package/BDFProxy.py", line 16, in <module>
    from core.widgets.docks.dockmonitor import (
  File "/usr/share/WiFi-Pumpkin/core/widgets/docks/dockmonitor.py", line 1, in <module>
    from plugins.extension import *
  File "/usr/share/WiFi-Pumpkin/plugins/extension/beef.py", line 1, in <module>
    from mitmproxy.models import decoded
ImportError: No module named mitmproxy.models

any ideas on why this aint working.
Thanks
ACE

101zombie101, 2019-01-21:
Is everything working yet?  Have had a chance to try it.

zer04you, 2019-11-20:
tray
sudo apt update
sudo apt install git isc-dhcp-server
git clone https://github.com/P0cL4bs/WiFi-Pumpkin.git
cd WiFi-Pumpkin
chmod +x installer.sh
sudo ./installer.sh --install
sudo pip install --upgrade pip
sudo pip install BeautifulSoup4 service_identity