P0cL4bs/WiFi-Pumpkin-deprecated #449: there is a problem

hexhunter404, 2019-01-04:
Before opening a new issue, please read this [wiki](https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Bug-reporting)

## What's the problem (or question)?

Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/package/TCP-Proxy.py", line 222, in run
    self.main()
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/package/TCP-Proxy.py", line 265, in main
    q = Queue.Queue()
AttributeError: class Queue has no attribute 'Queue'
[*] Pumpkin-Proxy running on port:8080 

nothing appears at monitor screen 
[image]

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): yes
* Version used tool: latest 
* Virtual Machine (yes or no and which): yes virtual box 6
* Operating System and version: Kali linux latest ver

mh4x0f, 2019-01-09:
nice, report I'll check this coming soon

m1sf1t42, 2019-01-14:
Hey there, 
found this on the python doc: https://docs.python.org/2/library/queue.html

In Python3 the Queue lib was renamed to 'queue'. I think the problem can be solved with the 2to3 Python converter. Works for me at least, running on a Lenovo ThinkPad L540, with the newest kali dist.

closed 2019-01-24 by mh4x0f

hdawg12, 2019-12-21:
Did you ever get a solution to this problem?  I also have the same error.  It works , but I don't see my AP, I created,  on the network.
This is my error:

[image]