P0cL4bs/WiFi-Pumpkin-deprecated #450: Nothing display in http-request monitor

N0cturn1s, 2019-01-04:
Hello Guyz,

Happy new year to all of you !
I installed wifi-pumpkin v0.8.7 on my kali.
The AP is working fine (I got internet acces on my client) but I can't see anything in the http-request / http-credential monitor :(
I've uninstalled v0.8.7 and forced v0.8.5 installation that I used to use some times ago and it use to work like a charm... but now i got the same probleme... nothing appear in the monitor section... my client get the ip adress and got acces to the internet... I'm pretty sure i miss something in the settings... (I tried with wifi-pumpkin plugin).
Thank you :)

closed 2019-01-09 by mh4x0f

mh4x0f, 2019-01-11:
look, I think found the problem #449 . I'll solve this coming soon

N0cturn1s, 2019-01-14:
@mh4x0f Ok thank you. It really seems the same yes. I got exactly the same messages.
Tried on two différent laptops (a Dell 5480 and a vostro 3350) with the same kali version.

If needed :
**Please tell us details about your environment.**

Card wireless adapters name (please check if support AP/mode): yes (Wireless panda PAU09
Version used tool: 0.8.7
Virtual Machine (yes or no and which): no
Operating System and version: Kali linux latest ver