P0cL4bs/WiFi-Pumpkin-deprecated #451: ARP Poisoner Issues Debian 9

thereporter42, 2019-01-06:
Before opening a new issue, please read this [wiki](https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Bug-reporting)

## What's the problem (or question)?

ARP Poisoning not functioning on Debian 9 

When I have it running Wifi Pumpkin running
as root  and I start it from the console and then I 
go to Modules and then to Arp Poisoner 
it is spitting out the following error.

Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 534, in show_arp_posion
    self.Farp_posion = GUIModules.frm_Arp_Poison(self.window_phishing)
  File "/usr/share/WiFi-Pumpkin/modules/poisoners/arp_poisoner.py", line 42, in __init__
    self.GUI()
  File "/usr/share/WiFi-Pumpkin/modules/poisoners/arp_poisoner.py", line 145, in GUI
    self.ConfigureEdits()
  File "/usr/share/WiFi-Pumpkin/modules/poisoners/arp_poisoner.py", line 174, in ConfigureEdits
    self.txt_mac.setText(Refactor.getHwAddr(x['activated'][0]))
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 237, in getHwAddr
    info = ioctl(s.fileno(), 0x8927,  pack('256s', ifname[:15]))
struct.error: argument for 's' must be a string

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: 
* Virtual Machine (yes or no and which): 
* Operating System and version: 

I am running Wifi-Pumpkin 0.8.7

lspci
02:00.0 Network controller: Qualcomm Atheros AR9285 Wireless Network Adapter (PCI-Express) (rev 01)
08:00.0 Ethernet controller: Qualcomm Atheros AR8152 v1.1 Fast Ethernet (rev c1)

And yes the wireless card supports AP that isn't the issue here the
AP is starting up fine but the arp poisoner is not working here 

The wireless card is the AR9285 Wireless Network Adapter
Wired network card is Qualcomm Atheros AR8152 v1.1 Fast Ethernet

This is Debian 9 
uname -a
Linux none 4.9.0-8-amd64 #1 SMP Debian 4.9.130-2 (2018-10-27) x86_64 GNU/Linux

python -V
Python 2.7.13

No Virtual machine 

And yes I ran the installer.sh and it installed all the needed dependencies 
Wifi-Pumpkin starts up just fine but the arp poisoner doesn't
appear to be working properly. 

I am not sure what version of Python you wrote that for but for 
the latest version it does not appear that it is working. 

Can you please take a look at this and fix this. If you need
more information please just reply and I will get it
for you don't close the ticket!  This issue needs to be 
addresed.

thereporter42, 2019-01-06:
Also if i run pip install -r requirements.txt 
all requirements are met! 

pip install -r requirements.txt

Requirement already satisfied: Twisted in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 1))
Requirement already satisfied: scapy in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 2))
Requirement already satisfied: bs4 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 3))
Requirement already satisfied: netaddr in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 4))
Requirement already satisfied: config in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 5))
Requirement already satisfied: dnspython in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 6))
Requirement already satisfied: isc_dhcp_leases in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 7))
Requirement already satisfied: netifaces in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 8))
Requirement already satisfied: pcapy in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 9))
Requirement already satisfied: configparser==3.3.0r1 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 10))
Requirement already satisfied: NetfilterQueue in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 11))
Requirement already satisfied: configobj in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 12))
Requirement already satisfied: libarchive-c==2.1 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 13))
Requirement already satisfied: python-magic==0.4.6 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 14))
Requirement already satisfied: pefile in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 15))
Requirement already satisfied: capstone in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 16))
Requirement already satisfied: hyperframe in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 17))
Requirement already satisfied: h2 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 18))
Requirement already satisfied: scapy_http in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 19))
Requirement already satisfied: service_identity in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 20))
Requirement already satisfied: beautifulsoup4 in /usr/lib/python2.7/dist-packages (from bs4->-r requirements.txt (line 3))
Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (from isc_dhcp_leases->-r requirements.txt (line 7))
Requirement already satisfied: future in /usr/local/lib/python2.7/dist-packages (from pefile->-r requirements.txt (line 15))
Requirement already satisfied: enum34<2,>=1.0.4; python_version == "2.7" or python_version == "3.3" in /usr/lib/python2.7/dist-packages (from h2->-r requirements.txt (line 18))
Requirement already satisfied: hpack<3,>=2.2 in /usr/local/lib/python2.7/dist-packages (from h2->-r requirements.txt (line 18))
root@none:/usr/local/pumpkin# QDBusConnection: session D-Bus connection created before QCoreApplication. Application may misbehave.
QDBusConnection: session D-Bus connection created before QCoreApplication. Application may misbehave.

thereporter42, 2019-01-06:
Also if i run pip install -r requirements.txt
all requirements are met!

pip install -r requirements.txt

Requirement already satisfied: Twisted in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 1))
Requirement already satisfied: scapy in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 2))
Requirement already satisfied: bs4 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 3))
Requirement already satisfied: netaddr in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 4))
Requirement already satisfied: config in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 5))
Requirement already satisfied: dnspython in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 6))
Requirement already satisfied: isc_dhcp_leases in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 7))
Requirement already satisfied: netifaces in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 8))
Requirement already satisfied: pcapy in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 9))
Requirement already satisfied: configparser==3.3.0r1 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 10))
Requirement already satisfied: NetfilterQueue in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 11))
Requirement already satisfied: configobj in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 12))
Requirement already satisfied: libarchive-c==2.1 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 13))
Requirement already satisfied: python-magic==0.4.6 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 14))
Requirement already satisfied: pefile in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 15))
Requirement already satisfied: capstone in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 16))
Requirement already satisfied: hyperframe in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 17))
Requirement already satisfied: h2 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 18))
Requirement already satisfied: scapy_http in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 19))
Requirement already satisfied: service_identity in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 20))
Requirement already satisfied: beautifulsoup4 in /usr/lib/python2.7/dist-packages (from bs4->-r requirements.txt (line 3))
Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (from isc_dhcp_leases->-r requirements.txt (line 7))
Requirement already satisfied: future in /usr/local/lib/python2.7/dist-packages (from pefile->-r requirements.txt (line 15))
Requirement already satisfied: enum34<2,>=1.0.4; python_version == "2.7" or python_version == "3.3" in /usr/lib/python2.7/dist-packages (from h2->-r requirements.txt (line 18))
Requirement already satisfied: hpack<3,>=2.2 in /usr/local/lib/python2.7/dist-packages (from h2->-r requirements.txt (line 18))

mh4x0f, 2019-01-06:
nice **report**, @thereporter42  I 'll fix this coming soon.

bedfl2016, 2019-01-21:
I also get a problem with mitmproxy.modules

mh4x0f, 2019-03-08:
thank's for report ;) @thereporter42

closed 2019-03-08 by mh4x0f