P0cL4bs/WiFi-Pumpkin-deprecated #453: [Feature Request] AppImage

zero77, 2019-01-08:
Can you please provide an AppImage, this is basically just a standalone portable executable for linux, that doesn't require installation.
Thanks

There are different ways to create AppImages:

    Use Open Build Service (OBS)
    Convert existing binary packages (.deb, .rpm, …)
    Bundling your Travis CI builds as AppImages
    Run linuxdeployqt on your Qt application
    Using electron-builder for Electron-based apps
    Create an AppDir manually

Documentation:
https://docs.appimage.org/packaging-guide/ways.html
https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages

mh4x0f, 2019-01-27:
hi, I read the doc and I don't found any example using the pyqt4, also  friend google unsuccessful.

zero77, 2019-01-28:
I am finding the same there's not so much info on python appimages.

This seems the best way to build an appimage (using Travis CI) but this example is for c, but it can be easley modified.
> https://github.com/probonopd/linuxdeployqt#using-linuxdeployqt-with-travis-ci
&
 https://www.youtube.com/watch?v=UJJKnr-WO0Y

This is what i have found so far but, it may be worth contacting an appimage developer for advice.

> https://github.com/AppImage/AppImageKit/issues/131
https://github.com/linuxdeploy/linuxdeploy-plugin-conda
https://github.com/AppImage/AppImageKit/wiki/Bundling-Python-apps

But i will update if i find any more info.
Thanks

zero77, 2019-01-29:
Theirs also:
pkg2appimage
https://docs.appimage.org/packaging-guide/converting-binary-packages/pkg2appimage.html?highlight=python

https://github.com/AppImage/AppImages/blob/master/pkg2appimage

mh4x0f, 2019-01-29:
all the links that you sent to me not help much. the ideal is find a app open source written in pyqt on appimage source.

zero77, 2019-02-01:
@mh4x0f 
Would converting a deb to an Appimage be an option because, theres is already a deb.

https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages#2-convert-existing-binary-packages