P0cL4bs/WiFi-Pumpkin-deprecated #454: Failure during install on raspberry pi 3+

jorgeguerron, 2019-01-08:
Before opening a new issue, please read this [wiki](https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Bug-reporting)

##Failure with lxml==3.6.0 and cryptography=1.5.3 What's the problem (or question)?
Not build wheel for lxml and cryptography, so this generate warnings because many methods have warnings and deprecated.

I intent downgrade the package but this not function.

I use kali-linux-2018.4a-rpi3-nexmon-64.img.xz image with raspberry pi 3 b+ and 64 gb sd card

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: 0.8.7
* Virtual Machine (yes or no and which): no, 
* Operating System and version: Kali Linux 2019.1

mh4x0f, 2019-01-09:
this is a problem with mitmproxy plugin, I'm trying to find a solution #440

closed 2019-01-11 by mh4x0f

jorgeguerron, 2019-01-11:
Dear Marcos, do you have any idea to update this? I use Wifi-pumpkin in
Debian on my laptop and don't have problem, and a month ago I use the tool
on my raspberry

mh4x0f, 2019-01-11:
@jorgeguerron  you need to reinstall the tool for install the version correct the mitmproxy