P0cL4bs/WiFi-Pumpkin-deprecated #457: mitmproxy.models

swucts, 2019-01-13:
when am trying to install wifi pumpkin i faced this issue and i've searched alot and find nothing when im trying to run it after installation when im writing sudo wifi-pumpkin i got that "No module named mitmproxy.models"
and i tried to do this commands but nothing again :- 
pip install mitmproxy
pip3 install mitmproxy
-----------------------------------

* Card wireless adapters name IEEE 802.11
* Version used tool: v0.8.7
* Virtual Machine :  Not virtual (linux mint)
* Operating System and version: Linux Mint
* Python versions: 2.7.15, 3.6.8 installed

Tusharrule, 2019-01-14:
Same here.
too annying.
Why not developer upgrade the mitm version

swucts, 2019-01-14:
i don't know where's the issue to solve it actually !

closed 2019-01-21 by mh4x0f