P0cL4bs/WiFi-Pumpkin-deprecated #466: Wifi-Pumpkin Post Installation Issue

mohbra, 2019-01-29:
Before opening a new issue, please read this [wiki](https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Bug-reporting)

## What's the problem (or question)?
I have got a Parrot OS running on my virtualbox and I added WiFi-Pumpkin but i got the error message below after installation
**You do not have a working installation of the service_identity module: 'cannot import name opentype'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.**

and on the WiFi-Pumpkin GUI under pumpkin proxy I get this

the package requirement mitmproxy==0.18.2 is not satisfied

I have gone through all the answers here with NO luck

*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
ALFA NETWORK 
[image]

* Version used tool: 
* Virtual Machine (yes or no and which): 
YES VirtualBox
* Operating System and version:

mh4x0f, 2019-01-29:
the mitmproxy package is the problem, I added support run WP without this package because I don't found a solution.

easyfic, 2019-02-01:
fixed the service_identity error after upgrading pyasn1-modules

pip install --upgrade pyasn1-modules

but "the package requirement mitmproxy==0.18.2 is not satisfied" remains in WP

3th3r3al3, 2019-02-01:
Same here on ParotOS.... and can only get it running on Kali live after installing LionSecs/Xerosploit which has most the proper packages.... this is really frustrating. As I really like how smooth ParotOS is... but still have to run Kali for both pr

DRAHCORE, 2019-02-07:
If you run Mitmproxy you could have this message : 
ImportError : cannot import ssl_match_hostname
So : 
# sudo apt-get install python-backports.ssl-match-hostname
After WP can running without the message not satisfied

minhducsun2002, 2019-02-18:
@mh4x0f is it possible to include `mitmproxy` itself in the package? That's a way to solve things out beside relying on distribution packages.

mh4x0f, 2019-02-18:
@minhducsun2002 no, the package mitmproxy need specific version the cryptography lib this is a problem.

minhducsun2002, 2019-02-20:
@mh4x0f If you can get `cryptography` to work, maybe including the library itself with https://github.com/P0cL4bs/WiFi-Pumpkin itself is an idea?

mh4x0f, 2019-02-20:
@minhducsun2002 you do not understand, even if the mitmproxy package is in the WP repo it will be compiled and will result in same errors.

minhducsun2002, 2019-02-20:
@mh4x0f I mean, you include `cryptography` itself, not just `mitmproxy`. It also breaks?

mh4x0f, 2019-02-20:
@minhducsun2002 yes

minhducsun2002, 2019-02-20:
I don't really know how on earth can I still invoke WiFi-Pumpkin normally even when rolling back to 7f5633f6cf033583b55c4b4181412c6be495bc26. I only have `mitmproxy` installed from Debian Stretch's official repository, and no appropriate version of `cryptography` (I have `2.3` whereas `<=1.6` is required).

Diabinhodo69, 2019-02-20:
Hello, 
I have something similar but I don't know if it is related (or if it can help the community). When I install pyasn1 0.4.1 I don't get errors when I start wifi-pumpkin but it warns about mitmproxy's pyasn1 version. When I install pyasn1 0.1.9 I get the same errors (as OP) when I start wifi-pumpkin and it warns about pyasn1-modules 0.2.4 butno warnings about mitmproxy.
[image]

Maybe it helps to workout something
Cheers

mh4x0f, 2019-03-12:
hi all, maybe the solution to install the mitmproxy is this:
``` sh
sudo apt-get install libssl1.0
sudo pip install mitmproxy==0.18.2
```
execute this command your system and check if mitmproxy module is installed successfully

minhducsun2002, 2019-03-12:
Okay, in short, you install the old library that was needed for compilation?

hakonamatata, 2019-03-14:
> 
> 
> hi all, maybe the solution to install the mitmproxy is this:
> 
> ```shell
> sudo apt-get install libssl1.0
> sudo pip install mitmproxy==0.18.2
> ```
> 
> execute this command your system and check if mitmproxy module is installed successfully

Tried this and **pip list** does show mitmproxy 0.18.2, but I still get the error...

kimocoder, 2019-03-14:
Same here.. had it for some time now

minhducsun2002, 2019-03-14:
@hakonamatata Try firing up a Python 2 interpreting instance and import `mitmproxy`. Does it work?

_my system has pretty weird behavior - WiFi-Pumpkin launches even when I roll it back to a version that still requires_ `mitmproxy` _to start up._

mh4x0f, 2019-03-14:
@hakonamatata which your system ?

hakonamatata, 2019-03-14:
> 
> 
> @hakonamatata which your system ?

Headless Raspberry Pi with "Kali Linux RaspberryPi 2 and 3 - 2019.1"

mh4x0f, 2019-03-14:
I think that the better option is rewrite the Pumpkin Proxy and remove the mitmproxy from dependencies. I'll start work this coming soon

hakonamatata, 2019-03-15:
> 
> 
> I think that the better option is rewrite the Pumpkin Proxy and remove the mitmproxy from dependencies. I'll start work this coming soon

Sounds like a good plan. 

btw. Why do we use such an old version of mitmproxy anyway? The current version is 4.0.4, and WiFi-Pumpkin uses version 0.18.2? It's a bit strange

minhducsun2002, 2019-03-15:
@hakonamatata `0.18.2`, as far as I know, is the last version of `mitmproxy` that supports Python 2.

And our framework uses Python 2.

I suppose you know the existence of [this](https://github.com/P0cL4bs/WiFi-Pumpkin/projects/3)?

JoeMarshall123, 2019-03-18:
any update ? 

I am using  Raspberry Pi 3 with Kali and when I attempted all of what was suggested above I get this: 
  File "/usr/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 309, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: Error([('SSL routines', 'ssl3_get_server_certificate', 'certificate verify failed')],)",),)': /simple/mitmproxy/  

and 

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 309, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
  Could not find a version that satisfies the requirement mitmproxy==0.18.2 (from versions: )
No matching distribution found for mitmproxy==0.18.2

Ideally i need mitmproxy to work.

ericng414, 2019-03-23:
Thanks All, I got the same problem and I fixed it with the following commands:
```
pip install --upgrade pyasn1-modules
apt-get install libssl1.0
pip install mitmproxy==0.18.2
apt-get install python-backports.ssl-match-hostname
```
I am using a native kali release 2019.1 and WP v0.18.2

SenexCrenshaw, 2019-03-30:
For me the fix was:

```
pip uninstall pyOpenSSL 
pip install pyOpenSSL mitmproxy==0.18.2
```
Raspberry Pi 3 B+ with Kali 2019.1 and WP git master

0b1w4n, 2019-04-06:
@ericng414 Thank you SOOOO much! 
That was the only solution that worked for me after hours of searching and multiple fails.
I am on ParrotOS 4.19 and WP v0.8.7

BiD1, 2019-05-03:
https://github.com/P0cL4bs/WiFi-Pumpkin/issues/466#issuecomment-475850943
Thanks All, I got the same problem and I fixed it with the following commands:

pip install --upgrade pyasn1-modules
apt-get install libssl1.0
pip install mitmproxy==0.18.2
apt-get install python-backports.ssl-match-hostname

I am using a native kali release 2019.1 and WP v0.18.2

This is work for me. i'm useing kali 2019.1 pumpkin v0.8.8
Tnx to ericng414

tiagojpavan, 2019-05-08:
> Thanks All, I got the same problem and I fixed it with the following commands:
> 
> ```
> pip install --upgrade pyasn1-modules
> apt-get install libssl1.0
> pip install mitmproxy==0.18.2
> apt-get install python-backports.ssl-match-hostname
> ```
> 
> I am using a native kali release 2019.1 and WP v0.18.2

Thanks man. This solved my problem with the mitmproxy on Parrot OS 4.6.

noobkid2000, 2019-05-09:
i tried all of them above but not solved yet ,i am using parrot 4.6 install on my pc , dual boot-- kali and patrrot, in kali perfectly it is working.
[image]
i got this error also
" Could not find a version that satisfies the requirement mitmproxy==0.18.2﻿ (from versions: )
No matching distribution found for mitmproxy==0.18.2﻿ "

hwac121, 2019-06-25:
> Thanks All, I got the same problem and I fixed it with the following commands:
> 
> ```
> pip install --upgrade pyasn1-modules
> apt-get install libssl1.0
> pip install mitmproxy==0.18.2
> apt-get install python-backports.ssl-match-hostname
> ```
> 
> I am using a native kali release 2019.1 and WP v0.18.2
--------------------------------------------------------------------------------------
Ok, tried that and it completely removed wifi pumpkin... LOL
Is this still being developed? I hate to see it like this as it is a really slick mitm app.
OH and those of you saying you have problems because of Parrot Security? I have ABSOLUTELY NO PROBLEMS running mitm apps on Parrot Security, which uses a lot of the kali repositories of software. Parrot is basically a tweaked out Kali. LOVE Parrot!

hwac121, 2019-06-25:
Install this version and problems go away and you will have new features too. [https://github.com/P0cL4bs/WiFi-Pumpkin](https://github.com/P0cL4bs/WiFi-Pumpkin)

hwac121, 2019-06-25:
Best version to install: [https://github.com/P0cL4bs/WiFi-Pumpkin](url)

TheGroundZero, 2019-06-27:
On Ubuntu the following worked:

As per @ericng414 and @SenexCrenshaw 

````
pip install --upgrade pyasn1-modules
apt-get install libssl1.1        # 1.1 instead of 1.0
pip uninstall pyOpenSSL 
pip install pyOpenSSL mitmproxy==0.18.2
apt-get install python-backports.ssl-match-hostname
````

Will complain about following requirements but finish nonetheless:

- cryptography<1.6,>=1.3 (installed = 2.3)
- pyOpenSSL<17.0,>=16.0 (installed 19.0.0)

sercanyilmaz84, 2019-08-27:
I have tried all recommends above but there is still same problem with Kali Linux.

vkpdeveloper, 2019-11-16:
> Thanks All, I got the same problem and I fixed it with the following commands:
> 
> ```
> pip install --upgrade pyasn1-modules
> apt-get install libssl1.0
> pip install mitmproxy==0.18.2
> apt-get install python-backports.ssl-match-hostname
> ```
> 
> I am using a native kali release 2019.1 and WP v0.18.2

Thanks, Bro It Works for me

sarmad-rauf, 2020-02-17:
I have tried all the above mentioned commands but my wifi-pumpkin is still showing the package requirement mitmproxy==0.18.2 is not satisfied I am running KALI linux. Please help me out!